jadonk/bonescript #106: bonescript module will not install on latest stable node v0.12.0

faddah, 2015-03-13:
hello,

i'm runniing node v0.12.0 and npm v2.7.1.

when attempting to install bonescript so i can start programming with my beagle board black rev. c, using ```$ npm install -g bonescript```, it will not install. i get the following —

```bash

$ npm install -g bonescript
npm WARN engine systemd@0.2.6: wanted: {"node":">=0.6.11 <0.11.0"} (current: {"node":"0.12.0","npm":"2.5.1"})
npm WARN engine serialport@1.2.5: wanted: {"node":"0.10.x"} (current: {"node":"0.12.0","npm":"2.5.1"})

> epoll@0.0.7 install /usr/local/.nvm/versions/node/v0.12.0/lib/node_modules/bonescript/node_modules/epoll
> node-gyp rebuild

child_process: customFds option is deprecated, use stdio instead.
make: Entering directory `/usr/local/.nvm/versions/node/v0.12.0/lib/node_modules/bonescript/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
In file included from ../src/epoll.cc:17:0:
../node_modules/nan/nan.h: In function ‘v8::Handle<v8::Value> NanError(const char*)’:
../node_modules/nan/nan.h:252:9: error: ‘New’ is not a member of ‘v8::String’
     fun(v8::String::New(errmsg))
         ^
../node_modules/nan/nan.h:261:12: note: in expansion of macro ‘_NAN_ERROR’
    return  _NAN_ERROR(v8::Exception::Error, errmsg);
            ^
../node_modules/nan/nan.h: In function ‘void NanThrowError(const char*)’:
../node_modules/nan/nan.h:257:7: error: ‘ThrowException’ is not a member of ‘v8’
       v8::ThrowException(_NAN_ERROR(fun, errmsg));                             \
       ^
../node_modules/nan/nan.h:265:5: note: in expansion of macro ‘_NAN_THROW_ERROR’
     _NAN_THROW_ERROR(v8::Exception::Error, errmsg);
     ^
../node_modules/nan/nan.h:252:9: error: ‘New’ is not a member of ‘v8::String’
     fun(v8::String::New(errmsg))
         ^
../node_modules/nan/nan.h:257:26: note: in expansion of macro ‘_NAN_ERROR’
       v8::ThrowException(_NAN_ERROR(fun, errmsg));                             \
                          ^
../node_modules/nan/nan.h:265:5: note: in expansion of macro ‘_NAN_THROW_ERROR’
     _NAN_THROW_ERROR(v8::Exception::Error, errmsg);
     ^
../node_modules/nan/nan.h: In function ‘void NanThrowError(v8::Handle<v8::Value>)’:
../node_modules/nan/nan.h:270:5: error: ‘ThrowException’ is not a member of ‘v8’
     v8::ThrowException(error);
     ^
../node_modules/nan/nan.h: In function ‘v8::Handle<v8::Value> NanError(const char*, int)’:
../node_modules/nan/nan.h:276:53: error: ‘New’ is not a member of ‘v8::String’
     v8::Local<v8::Value> err = v8::Exception::Error(v8::String::New(msg));
                                                     ^
../node_modules/nan/nan.h:278:14: error: ‘New’ is not a member of ‘v8::String’
     obj->Set(v8::String::New("code"), v8::Int32::New(errorNumber));
              ^
../node_modules/nan/nan.h:278:65: error: no matching function for call to ‘v8::Int32::New(const int&)’
     obj->Set(v8::String::New("code"), v8::Int32::New(errorNumber));
                                                                 ^
../node_modules/nan/nan.h:278:65: note: candidate is:
In file included from ../src/epoll.cc:13:0:
/home/chronos/user/.node-gyp/0.12.0/deps/v8/include/v8.h:2012:25: note: static v8::Local<v8::Integer> v8::Integer::New(v8::Isolate*, int32_t)
   static Local<Integer> New(Isolate* isolate, int32_t value);
                         ^
/home/chronos/user/.node-gyp/0.12.0/deps/v8/include/v8.h:2012:25: note:   candidate expects 2 arguments, 1 provided
In file included from ../src/epoll.cc:17:0:
../node_modules/nan/nan.h: In function ‘v8::Handle<v8::Value> NanTypeError(const char*)’:
../node_modules/nan/nan.h:252:9: error: ‘New’ is not a member of ‘v8::String’
     fun(v8::String::New(errmsg))
         ^
../node_modules/nan/nan.h:289:12: note: in expansion of macro ‘_NAN_ERROR’
     return _NAN_ERROR(v8::Exception::TypeError, errmsg);
            ^
[... 1263 lines cut ...]
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.
npm ERR! Linux 3.8.11
npm ERR! argv "/usr/local/.nvm/versions/node/v0.12.0/bin/node" "/usr/local/.nvm/versions/node/v0.12.0/bin/npm" "install" "-g" "bonescript"
npm ERR! node v0.12.0
npm ERR! npm  v2.5.1

npm ERR! Uncaught, unspecified "error" event.
npm ERR! 
npm ERR! If you need help, you may report this error at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /npm-debug.log

```

i had previously [filed this as an error with npm here - #7622](https://github.com/npm/npm/issues/7622), but was told by @othiym23 that this was, "Although I think node-serialport has been updated to work with the new V8 APIs in Node 0.12.0, bonescript is still depending on a version that's about 10 releases behind at this point." 

so i am filing this issue, seeing what needs updating so bonescript installs without errors via npm with node v0.12.0, please?  please advise.

best,

—  faddah
     portland, oregon, u.s.a.

AlexanderSelzer, 2015-04-01:
Same problem here. Not only is there a node version mismatch, but there are compiler errors everywhere.

 I guess the modules `epoll`, `i2c`, and `serialport` would have to be rewritten for it to work with node 0.12.

glowlabs, 2015-04-01:
Same issue here on multiple beaglebone black boards.

faddah, 2015-04-03:
one thought:  node v.0.12.0, while the latest stable version, has a number of issues, including with the npm serialport module. suggest backing down to 0.10.38, maybe, until things get fixed? 

best,

—  faddah
     portland, oregon, u.s.a.

arhodes18, 2015-09-17:
This thread looks really old now, but Im having the same issues here.

vladiks, 2015-10-07:
I have similar problem. I have Beagle Bone Black with Bone Debian 7.8. Installed current node and then tried install bonescript with no success and long list of errors.

bjgross1076, 2015-12-18:
Any plans to update?  I'm hoping to use beaglebone green with the Microsoft Azure IOT hub, but their script requries the latest version of node.

jadonk, 2016-02-10:
Please try the 0.5.x branch (0.5.0-beta1 just pushed to npm).

@bjgross1076 latest version of node or just newer?

kory44, 2016-05-10:
@jadonk  What is the highest node version that a version of bonescript supports? We are having issues with our embedded system because we coded in es6 which isn't running with node .10.

So,  which Node and which version of Bonescript, I mean.

jadonk, 2016-05-17:
0.12.x currently, via the 0.5.0 branch: npm install -g bonescript@0.5.0-beta6

We're working on enabling newer versions.

closed 2016-05-18 by jadonk