jadonk/bonescript #108: Support for Beaglebone devices running 4.1 kernel.

shaunmulligan, 2015-08-05:
Since the new capemgr has now been included in the latest 4.1 kernel, I was wondering if there was a plan to support it?

linux repo: https://github.com/beagleboard/linux/tree/4.1
overlay repo: https://github.com/RobertCNelson/bb.org-overlays

As far as I can see the major changes are in the file path
slots:
/sys/devices/platform/bone_capemgr/slots

I assume the BB foundation will be moving towards this kernel in the near future, so just interested in finding out if this lib will support it?

adityapatadia, 2015-08-05:
I can see you have also commented on OctalBoneScript. Yes, we are planning to support it and it will take a few months for us to upgrade OBS code.

jadonk, 2018-01-04:
For PWM, BoneScript now uses the pinmux helpers first introduced with cape-universal.

closed 2018-01-04 by jadonk