jadonk/bonescript #110: Pin Mux Error on P8_40

magicroomy, 2015-10-08:
Hi,
I stumbled ybout something that I would regard as an error in the pin-mux definition of P8_40
In bone.js, i can find the section for it.
  {
        "name": "GPIO2_13",
        "gpio": 77,
        "mux": "lcd_data7",
        "eeprom": 52,
        "key": "P8_40",
        "muxRegOffset": "0x0bc",
        "options": [
            "lcd_data7",
            "gpmc_a7",
            "pr1_edio_data_in7",
            "eqep2_strobe",
            "pr1_pru1_pru_r30_7",
            "pr1_pru_pru1_r30_7",
            "pr1_pru1_pru_r31_7",
            "gpio2_13"
        ]
    },

The following line indicate the modes for  mode 4-5
            "pr1_pru1_pru_r30_7",
            "pr1_pru_pru1_r30_7",
            "pr1_pru1_pru_r31_7",

I think it should be
            "pr1_edio_data_out_7",
            "pr1_pru1_pru_r30_7",
            "pr1_pru1_pru_r31_7",

Could you please check ?

jadonk, 2018-04-25:
This issue was moved to beagleboard/bonescript#34

closed 2018-04-25 by jadonk

vaishnav98, 2018-05-06:
@jadonk i looked at : http://www.ti.com/lit/ds/symlink/am3358.pdf , Page No:32 and found that the pin-mux definition of P8_40 should be as per the suggestion.Was the reference material the correct one , should i have looked for the information somewhere else?