jadonk/bonescript #114: Writing to PWM

smarquis, 2016-02-11:
Hi,

When i try to execute the following js code:
var b = require('bonescript');    
b.analogWrite('P8_34',0.7,200.0,console.log);

I get the following error code:

error: error updating PWM freq and value: undefined, Error: ENOENT, no such file or directory 'undefined/duty'
{ err: 'error updating PWM freq and value: undefined, Error: ENOENT, no such file or directory \'undefined/duty\'' }

I've installed fresh images, ran update, upgrades, and attempted all sort of work arounds. This problem shouldn't be persisting across completely fresh images. Is there a workaround to this?

Thanks for any help!

MarkAYoder, 2016-02-11:
I  there are two things happening here.  Have you disabled the HDMI?  If not, pin P8_34 is used by it.  I think that's why there is an 'undefined' in the path.

The second thing is a bug I fixed.  It turns out when the pwm is first used you have to change the period before changing the duty cycle.

This should be fixed now.  The fix is in my repo[1] waiting for Jason to pull it in.

--Mark

[1] https://github.com/MarkAYoder/bonescript

jadonk, 2016-02-14:
Merged into 0.5.x branch.

kory44, 2016-03-07:
I'm(noob) so confused, how do i get the version of bonescript on my BBB so that it will correctly do PWM?

like, what is the actual command? I have debian 7.9.

sputnik63, 2016-05-17:
If hw_oldkernel.js is used by bonescript maybe this issue could be caused by the issue #128

jadonk, 2016-05-18:
@smarquis Can I close this one?

smarquis, 2016-05-18:
Hi @jadonk 

You can close this issue. It has been resolved :) thanks @MarkAYoder for the help!

closed 2016-05-18 by jadonk