jadonk/bonescript #116: blinkled.js generates errors   error: Unable to find LED: usr0, error: Unable to find gpio: /sys/class/leds/beaglebone::usr3/brightness error: Unable to write to /sys/class/leds/beaglebone::usr3/brightness

akdbadoub, 2016-02-19:
i update bbb with the latest image debian 8.3 kernel 4.1 i run blinkled.js and i get the following errors:

error: Unable to find gpio: /sys/class/leds/beaglebone::usr0/brightness
error: Unable to write to /sys/class/leds/beaglebone::usr0/brightness
error: Unable to find gpio: /sys/class/leds/beaglebone::usr1/brightness
error: Unable to write to /sys/class/leds/beaglebone::usr1/brightness
error: Unable to find gpio: /sys/class/leds/beaglebone::usr2/brightness
error: Unable to write to /sys/class/leds/beaglebone::usr2/brightness
error: Unable to find gpio: /sys/class/leds/beaglebone::usr3/brightness
error: Unable to write to /sys/class/leds/beaglebone::usr3/brightness
error: Unable to write to /sys/class/gpio/gpio50/value
error: Unable to write to /sys/class/leds/beaglebone::usr0/brightness
error: Unable to write to /sys/class/leds/beaglebone::usr1/brightness
error: Unable to write to /sys/class/leds/beaglebone::usr2/brightness
error: Unable to write to /sys/class/leds/beaglebone::usr3/brightness
error: Unable to write to /sys/class/gpio/gpio50/value

MarkAYoder, 2016-02-20:
Hmmm.. the path should be /sys/class/leds/beaglebone:**green:**usr0/brightness.

It's working fine for me.  I'm running the image from https://rcn-ee.com/rootfs/bb.org/testing/2016-02-18/lxqt-4gb/

# **uname -a**
Linux beaglebone 4.1.17-ti-rt-r48 #1 SMP PREEMPT RT Fri Feb 12 23:46:00 UTC 2016 armv7l GNU/Linux
# **cat /etc/dogtag **
BeagleBoard.org Debian Image 2016-02-18

What are you running?

--Mark

akdbadoub, 2016-02-20:
uname -a
Linux beaglebone 4.1.15-ti-rt-r43 #1 SMP PREEMPT RT Thu Jan 21 20:13:58 UTC 2016 armv7l GNU/Linux

cat /etc/dogtag
BeagleBoard.org Debian Image 2016-01-24

btw re-imaged for 7.8 debian image and it worked fine.

MarkAYoder, 2016-02-20:
The BoneScript updates for 4.1 are recent.  Grabbing the newest image may fix your errors.

akdbadoub, 2016-02-21:
i just did that, and it worked. actually, down grading to 7.8 image also worked, so i guess that specific image is flowed thanks alot.

Workshopshed, 2017-02-28:
Ping, got the same issue:

debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[e066cb50c5eda24ce09227b1a155da823d56225a]
eeprom:[A335BNLTBLA21703EMBL0026]
dogtag:[BeagleBoard.org Debian Image 2017-02-19]
bootloader:[/dev/mmcblk1]:[U-Boot 2017.03-rc2-00002-g11d4fd]
kernel:[4.4.49-ti-r89]
nodejs:[v4.8.0]

jadonk, 2017-02-28:
@Workshopshed I'm able to reproduce the issue with the latest version of bonescript, but I'm **really** confused where it is coming from. Will debug soon.

jadonk, 2017-03-01:
One aspect of the issue is that hw_oldkernel.js gets used. It could be useful for me to fix that file.

For some newer kernels (4.4.x+) that don't have CapeMgr installed might also see this issue.

I've added a fix (62a1aadd3bfac536bcef42c3d0f6cd860ae1ac20) that simply forces hw_mainline.js to be used when the kernel is 4.4.x or newer. Those using 4.1 kernels should upgrade to newer kernels.

closed 2017-03-01 by jadonk