jadonk/bonescript #118: Can't install/upgrade bonescript

kory44, 2016-03-05:
I have the official debian 3.8 kernel. using some of the PWMs weren't working so i tried to upgrade the bonscript module. it failed, and now i can't use my old bonescript, and I can't install it either. 

here is my error message:

> ref@1.3.2 install /usr/local/lib/node_modules/bonescript/node_modules/ffi/node_modules/ref
> node-gyp rebuild

make: Entering directory `/usr/local/lib/node_modules/bonescript/node_modules/ffi/node_modules/ref/build'
  CXX(target) Release/obj.target/binding/src/binding.o
In file included from ../src/binding.cc:7:0:
../node_modules/nan/nan.h:328:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
make: *** [Release/obj.target/binding/src/binding.o] Error 1
make: Leaving directory `/usr/local/lib/node_modules/bonescript/node_modules/ffi/node_modules/ref/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 3.8.13-bone70
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /usr/local/lib/node_modules/bonescript/node_modules/ffi/node_modules/ref
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! ref@1.3.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the ref@1.3.2 install script.
npm ERR! This is most likely a problem with the ref package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls ref
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 3.8.13-bone70
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "-g" "bonescript"
npm ERR! cwd /root
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /root/npm-debug.log
npm ERR! not ok code 0

fivdi, 2016-03-06:
@kory44 it may be slightly out of date but see https://github.com/fivdi/onoff/wiki/Node.js-v0.10.29-and-native-addons-on-the-BeagleBone-Black

kory44, 2016-03-07:
Yes, above link worked.

do this first, before install:

apt-get update
apt-get upgrade

closed 2016-03-07 by kory44