jadonk/bonescript #122: PWM polarity is sometimes inverted.

kory44, 2016-04-17:
I have this error when initializing an object method that calls analogWrite. It causes my logic to be inverted so that 1 means off, and 0 means on. It only happens sometimes, so I can't simply change the logic.

self.Dim = function(amount){
			bone.analogWrite(self.pin, (amount));
			self.state = amount;
		}
		self.Dim(0.0);

> Sunset_dim = new GrowSwitch("P8_13",1,true);
error: Error enabling PWM controls: Error: ENOENT, no such file or directory '/sys/devices/ocp.3/bs_pwm_test_P8_13.21/polarity'
info: Error enabling PWM controls: Error: ENOENT, no such file or directory '/sys/devices/ocp.3/bs_pwm_test_P8_13.21/polarity'
error: error updating PWM freq and value: /sys/devices/ocp.3/bs_pwm_test_P8_13.21, Error: ENOENT, no such file or directory '/sys/devices/ocp.3/bs_pwm_test_P8_13.21/duty'
{ pin: 'P8_13',
  active: 1,
  isPWM: true,
  state: 0,
  Dim: [Function] }

jadonk, 2016-06-08:
Can you try performing a b.pinMode('P8_13', b.ANALOG_OUTPUT); ahead of making any b.analogWrite() attempts? Also, any chance you can try a 4.4.x kernel and 0.5.0-betaX BoneScript?

kory44, 2016-06-25:
setting pinMode did not work consistently. Is it a synchronous function? At this time I cannot try newer versions because it's taken some time to make everything else compatible. I don't want to break other things.

closed 2018-04-25 by jadonk