jadonk/bonescript #131: BoneScript PWM initialization error

khaled-hossain-code, 2016-06-04:
Here I am trying to turn on a buzzer for one second and then turn it off. Here is the code

`var b = require('bonescript');
var buzzer = "P8_19";
b.analogWrite(buzzer,  0.5, 2000);

setTimeout(function() {
    b.analogWrite(buzzer, 0);     // Turn off buzzer after milliseconds time 
    }, 1000);
`

This code gives the following error after i try to run it immediately after reboot
error: Error enabling PWM controls: Error: ENOENT, no such file or directory '/sys/devices/ocp.3/bs_pwm_test_P8_19.15/polarity'

I tried to set "P8_19" as output but it gives compile error. 

But after first attempt, this code runs smoothly without any problem. How to solve this problem?

jadonk, 2016-06-08:
@khaled-hossain-code would you object to trying the 4.4.x kernel and 0.5.0-betaX BoneScript?

khaled-hossain-code, 2016-06-09:
I am using this code in production, My rest of code is complete, so I am trying to carry on with the default beaglebone debian image and 3.8 kernel. is there any way to solve it in the default kernel and default bonescript?

jadonk, 2016-06-15:
Try using b.pinMode('P9_13', b.ANALOG_OUTPUT); to configure the pin at startup before calling b.analogWrite().

closed 2017-12-08 by khaled-hossain-code