jadonk/bonescript #134: kernel 4.1 doesn't use eeprom file, instead, it's called nvmem

jgtaylor, 2016-06-12:
My setup:
```
name: 'BeagleBoard.org BeagleBone Black\u0000',
  bonescript: '0.5.0-beta7',
  os: 
   { hostname: 'auto-garden',
     type: 'Linux',
     arch: 'arm',
     release: '4.1.25-ti-r62',
     uptime: 10719,
     loadavg: [ 0.16650390625, 0.1044921875, 0.06201171875 ],
     totalmem: 516202496,
     freemem: 300998656,
     networkInterfaces: { lo: [Object], eth0: [Object] } },
  version: '00C0',
  serialNumber: '4215BBBK177E',
  dogtag: 'BeagleBoard.org Debian Image 2016-03-20\n' }
```
getPlatform fails reading the eeprom data. In the latest image, the file is called:
/sys/bus/i2c/devices/0-0050/at24-0/nvmem  # rather than the eeprom filename used in the script.

```
> b.getPlatform();
TypeError: Cannot read property 'version' of undefined
  at Object.exports.readPlatform (/usr/local/lib/node_modules/bonescript/src/hw_mainline.js:297:25)
  at Object.f.getPlatform (/usr/local/lib/node_modules/bonescript/src/index.js:524:19)
  at repl:1:3
  at REPLServer.defaultEval (repl.js:262:27)
  at bound (domain.js:287:14)
  at REPLServer.runBound [as eval] (domain.js:300:12)
  at REPLServer.<anonymous> (repl.js:431:12)
  at emitOne (events.js:82:20)
  at REPLServer.emit (events.js:169:7)
  at REPLServer.Interface._onLine (readline.js:211:10)
  at REPLServer.Interface._line (readline.js:550:8)
  at REPLServer.Interface._ttyWrite (readline.js:827:14)
  at ReadStream.onkeypress (readline.js:106:10)
  at emitTwo (events.js:87:13)
  at ReadStream.emit (events.js:172:7)
  at emitKeys (readline.js:1251:14)
  at [object Generator].next (native)
  at ReadStream.onData (readline.js:919:36)
  at emitOne (events.js:77:13)
  at ReadStream.emit (events.js:169:7)
  at readableAddChunk (_stream_readable.js:153:18)
  at ReadStream.Readable.push (_stream_readable.js:111:10)
  at TTY.onread (net.js:531:20)
```
changing lines 284 & 286 resolve this specific issue, however, I don't think it's a permanent fix.

jgtaylor, 2016-06-14:
this appears to be an issue with changing device names between 4.1 kernel and 4.4.

jgtaylor, 2016-06-15:
w/ kernel "4.4.13-bone11", there are still additional differences.
Lines 284/286 (and other eeprom references) reference to the /sys filesystem are (as of this kernel), called: sys/bus/i2c/devices/2-0054/eeprom ( that is, remove the at24-x directory)

I confess, I'm a bit confused - are these paths set via kernel compile options, device tree snippets, somewhere in the code? I don't get how they're changing so much between ti-channel/bone-channel and the different numbered versions.

jadonk, 2016-06-15:
Check this on the 0.5.x branch.

jadonk, 2016-06-15:
They are set in the kernel source code. They change partially due to us using out-of-tree patches to enable these features. As they are accepted into mainline, deltas should slow down.

jadonk, 2017-03-01:
See 58b7def25f7159640393a9a7de28f95c5e475d68

Please reopen if you see this again.

closed 2017-03-01 by jadonk