jadonk/bonescript #136: pinGetMode(pin, callback()); RegEx error due to format change

jgtaylor, 2016-06-12:
old 3.8 kernel format:
pin 127 (44e109fc) 00000028 pinctrl-single

new 4.1 kernel format:
pin 127 (44e109fc.0) 00000028 pinctrl-single 

note the addition of ".0" in the muxAddress field.

change line 81: parse.js
   var pattern = new RegExp('pin ([0-9]+) .([0-9a-f]+\.[0-9]). ([0-9a-f]+) pinctrl-single'); // added "\.[0-9]"  

not sure if this will work for everything.
recreate by using the callback on getPinMode

```
var b = require('bonescript');
b.pinMode('P8_13', b.ANALOG_OUTPUT);
b.analogWrite('P8_13', 0.5, 10000);
b.getPinMode('P8_13', function(x) { console.log(x)});
```
returns:

```
> { pin: 'P8_13',
  name: 'EHRPWM2B',
  options: 
   [ 'gpmc_ad9',
     'lcd_data22',
     'mmc1_dat1',
     'mmc2_dat5',
     'ehrpwm2B',
     'pr1_mii0_col',
     'NA',
     'gpio0_23' ],
  pwm: {},
  gpio: { active: true, direction: 'in', allocated: false },
  mux: 4,
  slew: 'fast',
  rx: 'enabled',
  pullup: 'pulldown' }

>
``` 

Still an issue with the pwm: {} object. I'll open another issue for this.

jgtaylor, 2016-06-15:
this change is still required for kernel 4.4.13-bone11

jadonk, 2016-07-05:
Thank you very much for this bug report!

closed 2016-07-05 by jadonk