jadonk/bonescript #137: readPWMFreqAndValue() new kernel uses new names

jgtaylor, 2016-06-12:
for Kernel 4.1 (4.1.25-ti-r62 #1 SMP Tue May 24 21:23:52 UTC 2016 armv7l GNU/Linux)
modify lines 25 & 26 of hw_mainline.js:

```
//var period = fs.readFileSync(pwmPrefix[pin.pwm.name]+'/period_ns');                                                                             
var period = fs.readFileSync(pwmPrefix[pin.pwm.name]+'/period');                                                                                  
//var duty = fs.readFileSync(pwmPrefix[pin.pwm.name]+'/duty_ns');                                                                                 
var duty = fs.readFileSync(pwmPrefix[pin.pwm.name]+'/duty_cycle'); 
```

jadonk, 2016-06-15:
We are likely going to skip kernel 4.1 and go straight to supporting 4.4. See the 0.5.x branch.

jgtaylor, 2017-11-30:
Looking at the Readme.md in https://github.com/beagleboard/linux, it looks like 4.1 has fallen off the list, with 4.4, 4.9 & 4.14 remaining. Given the effort required to support so many different versions, I propose to close this PR.

closed 2017-11-30 by jgtaylor