jadonk/bonescript #144: ocp:P8_12_pinmux was not found

pdp7, 2016-10-12:
This is issue was reported on the Adafruit forums:
[Re: RE: Running Some Software and Making My BBB Go/Seth](https://forums.adafruit.com/viewtopic.php?f=49&t=103789#p523231)

This error occurred:
```
ocp:P8_12_pinmux was not found under /sys/devices/platform/ocp
```

I've been able to reproduce this error with the test code in the bonescript README:
```
root@beaglebone:~# cat bonescript-test.js 
var b = require('bonescript');

b.pinMode('P8_12', b.INPUT);
b.pinMode('P8_13', b.OUTPUT);

setInterval(copyInputToOutput, 100);

function copyInputToOutput() {
    b.digitalRead('P8_12', writeToOutput);
    function writeToOutput(x) {
        b.digitalWrite('P8_13', x.value);
    }
}
```

Error:
```
root@beaglebone:~# node ./bonescript-test.js 

/root/node_modules/bonescript/src/hw_mainline.js:84
    if(!pinmux) { throw p + " was not found under " + my.is_ocp(); }
                                                    ^
ocp:P8_12_pinmux was not found under /sys/devices/platform/ocp
```

My BeagleBone is running:
```
root@beaglebone:~# uname -a
Linux beaglebone 4.4.21-bone13 #1 Fri Sep 16 07:34:28 UTC 2016 armv7l GNU/Linux
root@beaglebone:~# cat /etc/debian_version 
8.6
root@beaglebone:~# cat /etc/dogtag 
BeagleBoard.org Debian Image 2016-08-14
root@beaglebone:~# node --version
v0.12.16
root@beaglebone:~# node -pe "require('bonescript').getPlatform().bonescript"
0.5.0

root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-SPIDEV1
 5: P-O-L-   1 Override Board Name,00A0,Override Manuf,am33xx_pwm

root@beaglebone:~# cat /boot/uEnv.txt  |grep -v ^# |grep -v '^$'
uname_r=4.4.21-bone13
cmdline=coherent_pool=1M quiet fbcon=map:10
cape_enable=bone_capemgr.enable_partno=BB-SPIDEV1
```

JamesHagerman, 2016-11-10:
I'm also having this issue. I'm running Ubuntu 16.04 on a BBB with the 4.4.28 kernel with bonescript 0.5.1

Any help on this would be great!

jadonk, 2016-11-11:
Guessing that this pin doesn't have a pinmux helper loaded. I'll try to check tomorrow, if not then, later next week.

e29qwg, 2016-11-22:
I also faced this issue. For a little workaround, you need to manually load device tree overlay by using following command on BeagleBone's shell.

```sh
config-pin overlay cape-universal
```

Since I'm using BB Green Wireless, my workaround is as followed.

```sh
config-pin overlay univ-bbgw
```

Hope this help

pdp7, 2016-11-24:
Thanks @e29qwg.  Yes, the problem was that I did not have cape-universal loaded.  It works now:
```
root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universaln
root@beaglebone:~# uname -a
Linux beaglebone 4.4.30-ti-r65 #1 SMP Tue Nov 8 17:28:39 UTC 2016 armv7l GNU/Linux
root@beaglebone:~# cat /etc/debian_version 
8.6
root@beaglebone:~# cat /etc/dogtag 
BeagleBoard.org Debian Image 2016-11-21
```

```
root@beaglebone:~# cat /boot/uEnv.txt  |grep -v ^# |grep -v '^$'
uname_r=4.4.30-ti-r65
cmdline=coherent_pool=1M quiet cape_universal=enable
```

```
root@beaglebone:~# node -pe "require('bonescript').getPlatform().bonescript"
0.5.2
```

closed 2016-11-24 by pdp7

Jim-Zenn, 2019-09-04:
@e29qwg I looked all over the internet, finally fixed my beaglebone with your command:

`config-pin overlay univ-bbgw`

Thank you T^T