jadonk/bonescript #145: Bonescript does not run with node.js v6.x

psiphi75, 2016-10-28:
Bonescript does not run with node.js v6.x (specifically v6.9.1).  Due to the `bindings` package within the `epoll` package, see this issue:  https://github.com/TooTallNate/node-bindings/issues/26

How to reproduce in bonescript:

    require('bonescript')

gives the following error:

```
Error: Could not locate the bindings file. Tried:
 → /root/node_modules/ref/build/binding.node
 → /root/node_modules/ref/build/Debug/binding.node
 → /root/node_modules/ref/build/Release/binding.node
 → /root/node_modules/ref/out/Debug/binding.node
 → /root/node_modules/ref/Debug/binding.node
 → /root/node_modules/ref/out/Release/binding.node
 → /root/node_modules/ref/Release/binding.node
 → /root/node_modules/ref/build/default/binding.node
 → /root/node_modules/ref/compiled/6.9.1/linux/arm/binding.node
    at bindings (/root/node_modules/bindings/bindings.js:88:9)
    at Object.<anonymous> (/root/node_modules/ref/lib/ref.js:5:47)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/root/node_modules/ffi/lib/ffi.js:6:11)

```

fivdi, 2016-11-06:
The stack trace indicates that the error is related to the `ref` module requiring the `bindings` module. There's no mention of the `epoll` module in the error message.

rogueturnip, 2016-11-08:
Has this been resolved? I'd like to move to node v7 on the beaglebone but need to keep bonescript working.

jadonk, 2016-11-11:
@rogueturnip @RobertCNelson does that mean that v7 is in nodesource? Shall I try that as my next version to intersect? Does Cloud9 IDE work with it?

RobertCNelson, 2016-11-11:
@jadonk for bonescript, 0.12.x -> v4.x, i2c needs to be bumped from 0.2.1 -> 0.2.2 (or 0.2.3 current master)..

here's the commits:
https://github.com/kelly/node-i2c/commit/6e00d8da5c4f79eb20aaf30f8ece8a3b9f56d987
https://github.com/kelly/node-i2c/commit/29b39fb63093b34165cfde3fdb5354eec05c554d

Note, this "will" break < v4.x, so you'll want to do a major bump..

Cloud9 IDE is currently built for, v0.12.x, v4.x, and v6.x

I'll have to review the other projects, but now is a great time to start migrating the stack to v4.x. (that's the default in Debian Stretch (9) at the moment)..

Regards,

jadonk, 2016-11-12:
So, do we need to reject this request to move to v7? Do we need to settle for v4 or can we jump to v6?

RobertCNelson, 2016-11-13:
I'll start the v4.x transition next Monday, once everything is working again, we will start on the v6.x transition. I'm hoping most of the fixes we'll need for v4.x should help v6.x...

rwaldron, 2016-11-14:
@jadonk (from left field!) If I may suggest: stick with support for only the LTS releases: https://github.com/nodejs/LTS#lts-schedule (that's what we're doing with Tessel 2 and so far everyone respects and appreciates the commitment)

RobertCNelson, 2016-11-15:
@jadonk for v4, we need one package bump, for v6, we need 3 bumps..

for v4+:
i2c : 0.2.1 -> 0.2.3
(this i2c bump will break v0.12.x)

then for v6+
serialport : 2.0.6 -> 3.1.2
transition docs: "Upgrading from 2.x to 3.x"
https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/UPGRADE_GUIDE.md

epoll : 0.1.17 -> 0.1.18

based on:
https://github.com/fivdi/epoll/commits/master
https://github.com/fivdi/epoll/commit/81564aba298e1839e5c8c5a2298c50223c3cd4a8

pre-builds are located here:

https://github.com/rcn-ee/npm-package-bonescript/tree/master/deploy

I'll be bumping the bonescript shortly, it'll detect which nodejs version v0.12.x/v4.x/v6.x via:

/usr/bin/nodejs --version

and extract the pre-build into /usr/local/lib/node_modules/

Regards,

fivdi, 2016-11-15:
@RobertCNelson  If `npm install bonescript` is called from the command line, the latest version of `bonescript` will be installed. Which version of `bonescript` will this be?  The version that works with Node.js v0.12, v4, or v6?

RobertCNelson, 2016-11-15:
@fivdi for our beagleboard user, we actually pre-build the npm package and just un-tar it in our custom deb packages..  Having the deb package call "npm install xyz" was just too un-reliable..

Right now:

```
npm install bonescript
```
is only v0.12.x due to the i2c dependency..

Regards,

bennyn, 2017-01-19:
Hey, I am new to the BeagleBone Community. I just got a [BeagleBone Black Wireless](https://beagleboard.org/black-wireless). What's the current status with running a more recent version of Node.js than v0.12.17 on the Beaglebone Black?

I can see that there were commits dedicated to run on Node.js v4.x:
- https://github.com/jadonk/bonescript/commit/ad374711ba1cf95cb95ffcfa883a7efad5ddac2a

What's missing right now to run BoneScript with Node.js v4.x? Because Node.js v.0.12 reached end of life on  2016-12-31 (https://github.com/nodejs/LTS#lts-schedule).

If it is about I2C, then we can maybe learn something from the [Johnny-Five project](http://johnny-five.io/) which [supports Node v4.2.1](http://johnny-five.io/#hello-world)?

RobertCNelson, 2017-01-19:
@bennyn so all the installed nodejs packages (except for one seeed tool in the seeed image) in our repo currently support v0.12/v4/v6.  So you could take the "latest" ( http://beagleboard.org/latest-images ) and change the nodesource repo from v0.12.x -> v4.x.. and just apt update ; apt upgrade...

Or just grab one of the latest weekly snapshots:

https://rcn-ee.net/rootfs/bb.org/testing/2017-01-15/

as v4.x is now set as default.. (really anything after 2016-11-06, has v4.x as default)

Regards,

bennyn, 2017-01-20:
Thanks @RobertCNelson, this helped a lot! 

I successfully upgraded my BBBW from Debian GNU/Linux 8.6 (jessie) with Node.js v0.12.17 to Debian GNU/Linux 8.7 (jessie) with Node.js v4.7.2. 🎉 

Now it's time to get my hands onto BoneScript... 😎

psiphi75, 2017-02-21:
I have lightly tested the latest code from Git on my BB and it seems to work with Node v4.x.  I'm going to test it further over the next couple of days.  When do you think v0.6.0 will make it to npm?

lorenx, 2017-10-20:
It perfectly works with:
- Debian Stretch
- node v6.11.4
- npm v3.10.10
- bonescript v0.6.1

Unfortunately, I just had almost the same _bindings_ error with Nodejs 8, any news about this?
Thank you very much for your work.

RobertCNelson, 2017-10-20:
@lorenx i'll setup a stretch/nodejsv8 build node tonight/weekend, and see where we are at..

RobertCNelson, 2017-10-20:
@lorenx it built on v8:
https://github.com/rcn-ee/npm-package-bonescript/blob/master/deploy/stretch/bonescript-0.6.1-c2065b3-v8.7.0.tar.xz

fivdi, 2017-10-22:
BoneScript v0.6.1 installs successfully for me with Node.js v8.7.0.
See https://gist.github.com/fivdi/27a4e8bb08953d8f3f8e8c7d3f563cab
There are lots of warnings but no errors.

A quick test also shows that it functions at runtime.

The following may look like a serialport error but it isn't:
```
node-pre-gyp http GET https://github.com/voodootikigod/node-serialport/releases/download/3.1.2/serialport-v3.1.2-node-v57-linux-arm.tar.gz
node-pre-gyp http 404 https://github.com/voodootikigod/node-serialport/releases/download/3.1.2/serialport-v3.1.2-node-v57-linux-arm.tar.gz
node-pre-gyp ERR! Tried to download: https://github.com/voodootikigod/node-serialport/releases/download/3.1.2/serialport-v3.1.2-node-v57-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@3.1.2 and node@8.7.0 (node-v57 ABI) (falling back to source compile with node-gyp) 
```
This just means that there are no pre-compiled serialport binaries available for download and that serialport should be compiled locally.

That being said, it should be possible to bump serialport to v4.0.7 and remain compatible with serialport v3.1.2. serialport v5 and higher are no longer compatible.

If versions are bumped I'd also suggest bumping from epoll v0.1.18 to v1.0.0. epoll v1.0.0 supports Node.js versions 0.10, 0.12, 4, 5, 6, 7 and 8.

jadonk, 2017-11-29:
@RobertCNelson any updates on where we are at on versions? I had to remove some really old versions of node.js from travis-ci recently due to my adding some usage of some newer language syntax to make a copy of an object. (https://github.com/jadonk/bonescript/commit/28ed000ec2818806343191ff69193a68a3fab736 https://travis-ci.org/jadonk/bonescript/jobs/300952499#L837)

If we are cool moving to at least nodejs v4, we can probably bump some of the dependencies. Lots of testing needed.

RobertCNelson, 2017-11-29:
@jadonk 

Requiring >= v4 would be a good bump.

From my notes:

node-red requires >= v4

Currently we are shipping:

Jessie: v4
Stretch: v6
buster: v8 (and then v10 when we get closer)

https://github.com/nodejs/Release

Regards,

closed 2017-12-20 by jadonk