jadonk/bonescript #146: input pin p8_44 to p8_41 and p8_31 issue on debian 8.x

udoom, 2016-12-26:
hello,
I use official debian images for my BBB.
And since debian 8.4+ those pin : p8_44 to p8_41 and p8_31, does not do their job as INPUT pin.
they can not fire CHANGE event anymore : 
```
    b.pinMode(io, b.INPUT, 7, 'pulldown');
    b.attachInterrupt(io, true, b.CHANGE, function(x){
        console.log(io);
    });
```

I have tested it on 8.4 and 8.6 debian. same issue.
but on 7.9 debian it works like a charm.

I also have tried it with original bonscript pkg on debian images and after a update to 0.5-rc

did you have plan to fix this ?
thank's a lot

fivdi, 2017-09-30:
[this](https://github.com/jadonk/bonescript/issues/154#issuecomment-333335431) may help here.

udoom, 2017-10-21:
hello,
yes I have updated all my /boot/uEnv.txt to disable video : 
disable_uboot_overlay_video=1
my dmesg is right: I can see all the : 
`gpio-of-helper ocp:cape-universal: Allocated GPIO id=****`

in fact, the problem is more simple : all pins listed (P8_41 to 44 & P8_31) have always their state=1.
here you can see my test.js script : 
(P8_41 is connected to nothing)

```
var b = require('bonescript');
var io = 'P8_41';
b.pinMode(io, b.INPUT);
setInterval(function() {
	b.digitalRead(io, function printStatus(x) {
		console.log('x.value = ' + x.value);
		console.log('x.err = ' + x.err);
	});
}, 1000);
```
and js console says : 

```
4:4:91-ti-r133
x.value = 1
x.err = undefined
x.value = 1
x.err = undefined
```

I have tested that script on 2 BBB with "[debian strech iot 9.2](https://debian.beagleboard.org/images/bone-debian-9.1-iot-armhf-2017-09-21-4gb.img.xz)" (latest bbb image), and same result.
if I try that on another pin (like P8_33) it works (state is 0).
I try that on oldest distribution ([debian 7.11](https://debian.beagleboard.org/images/bone-debian-7.11-lxde-4gb-armhf-2016-06-15-4gb.img.xz))
did you reproduce the issue ?

thank's

fivdi, 2017-10-22:
I see the same behavior. However, I don't think it's a bug. If nothing is connected to P8_41 and the program from your second post that prints the state of the pin every second is run, then it's not reasonable to assume that the state of the pin should be 0.

If nothing is connected to the pin and neither the internal pull-up resistor nor the internal pull-down resistor for the pin is activated, and in this case neither of them are activated, then the pin is floating and can be in any state. In this case the state of the pin just happens to be 1.

The file `/sys/devices/platform/ocp/ocp:P8_41_pinmux/state` can be used to control the pin mode for P8_41. When your program is running this file contains the string `gpio`. The internal pull-down resistor can be activated by writing `gpio_pd` to this file.

```
echo gpio_pd > /sys/devices/platform/ocp/ocp:P8_41_pinmux/state
```

To the best of my knowledge BoneScript does not provide any means for activating pull-up or pull-down resistors. In the past it may have provided this functionality.

Can you not use an external pull-down resistor to prevent the pin from floating?

udoom, 2017-10-22:
you are right: pulldown mode seems not be activated by default on debian 8.4+ (but it is on debian 7.11).
if I patch /usr/local/lib/node_modules/bonescript/src/hw_mainline.js on line 87 with : 
`fs.writeFileSync(pinmux+"/state", 'gpio');`
it works

the [pinMode bonescript api](http://beagleboard.org/Support/BoneScript/pinMode/) references that : 
```
pin: the BeagleBone pin identifier
direction: INPUT, INPUT_PULLUP (under development on 3.8 kernel) or OUTPUT
mux: index to mux mode (under development on 3.8 kernel)
pullup: 'pullup', 'pulldown' or 'disabled' (under development on 3.8 kernel)
slew: 'fast' or 'slow' (under development on 3.8 kernel)
callback: called upon completion
```

and even if I try that test.js script, it does not work : 
```
var b = require('bonescript');
var io = 'P8_41';
b.pinMode(io, b.INPUT, 7, 'pulldown');
setInterval(function() {
	b.digitalRead(io, function printStatus(x) {
		console.log('x.value = ' + x.value);
		console.log('x.err = ' + x.err);
	});
}, 1000);
```

dont'you assume that if it works on debian 7.11 and not on 8.4+, it is a regression ?

fivdi, 2017-10-22:
> don't you assume that if it works on debian 7.11 and not on 8.4+, it is a regression ?

That sounds like a reasonable assumption to make.

closed 2017-12-20 by jadonk