jadonk/bonescript #154: EACCES: permission denied, open '/sys/devices/platform/ocp/ocp:P9_31_pinmux/state'

udoom, 2017-08-26:
hi,
I have just installed new debian 9.1 iot on my bbb.
and I found this issue when my bonescript try to use p9_31 as GPIO_OUT : 

Error: EACCES: permission denied, open '/sys/devices/platform/ocp/ocp:P9_31_pinmux/state'
  at Error (native)
  at Object.fs.openSync (fs.js:641:18)

/sys/devices/platform/ocp/ocp:P9_31_pinmux/state is missing
(but P9_41 is present for example)

# uname -a
Linux beaglebone 4.4.83-ti-r119 #1 SMP Thu Aug 17 06:39:24 UTC 2017 armv7l GNU/Linux
# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universala

did you have a workaround ?
thank's

fivdi, 2017-09-02:
Here's a picture of the BeagleBone Black expansion headers take from the [BeagleBone 101](http://beagleboard.org/Support/bone101/).
[image]
If the Debian image is used with the default configuration then only the pins that are colored light green (Available Digital) and dark green (Available PWM) can be used as digital IOs. (An exception to this rule is P9_25 which shouldn't be colored light green and can't be used as it's reserved for audio if I remember correctly.)

udoom, 2017-09-03:
hello,
yes I know that, but if you load universala cape, you can activate all gpio on bbb, and you will have this configuration : 
[http://beagleboard.org/static/images/cape-headers-digital.png](url)

as you can see in my first post : 

cat /sys/devices/platform/bone_capemgr/slots
0: PF---- -1
1: PF---- -1
2: PF---- -1
3: PF---- -1
4: P-O-L- 0 Override Board Name,00A0,Override Manuf,cape-universala

fivdi, 2017-09-30:
Apologies for the delay in answering.

If a recent version of Debian is being used the pins that are normally reserved for HDMI video, for example P9_31, can be freed for usage as GPIOs by adding the following line to `/boot/uEnv.txt`:
```
disable_uboot_overlay_video=1
```
This is documented [here](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Disable_on-board_devices).

Due to a recent change `/sys/devices/platform/bone_capemgr/slots` is disabled after U-Boot has taken care of overlays. See [here](https://github.com/beagleboard/linux/issues/139).

udoom, 2017-10-14:
Hi,
thank's for your answer.
I tried to use uboot like you say : 

enable_uboot_overlays=1
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
enable_uboot_cape_universal=1

but I get this error, when I try to set p9_41 as GPIO as INPUT : 

/usr/local/lib/node_modules/bonescript/src/hw_mainline.js:84
    if(!pinmux) { throw p + " was not found under " + my.is_ocp(); }
                  ^
ocp:P9_41_pinmux was not found under /sys/devices/platform/ocp

I just flashed my BBB with debian_stretch_9.2_iot

thank's for your help

fivdi, 2017-10-14:
I can't reproduce the error.

The following program successfully blinks an LED connected to P9_41 with a frequency of 1Hz:
```js
var b = require('bonescript');
var ledState = b.LOW;

b.pinMode('P9_41', b.OUTPUT);

setInterval(function () {
  ledState = ledState === b.LOW ? b.HIGH : b.LOW;
  b.digitalWrite('P9_41', ledState);
}, 500);
```

The following program successfully prints 0 (the state of P9_41) to the screen:
```js
var b = require('bonescript');

b.pinMode('P9_41', b.INPUT);
console.log(b.digitalRead('P9_41'));
```

Please post a complete program that can be used to reproduce the error.

The complete content of `/boot/uEnv.txt` is:
```
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.4.91-ti-r133
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
```

udoom, 2017-10-14:
I can't understand : 

I flash my sdcard from skratch with debian iot 9.2, again.
I just tune /boot/uEnv.txt like yours : 

```
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.4.91-ti-r133
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
```

I create a test.js file in cloud9 root dir with your test content : 

```
var b = require('bonescript');
b.pinMode('P9_41', b.INPUT);
console.log(b.digitalRead('P9_41'));
```

and I've got in cloud9 console  : 

```
4:4:91-ti-r133

/usr/local/lib/node_modules/bonescript/src/hw_mainline.js:84
    if(!pinmux) { throw p + " was not found under " + my.is_ocp(); }
                  ^
ocp:P9_41_pinmux was not found under /sys/devices/platform/ocp
```

what's wrong ?

fivdi, 2017-10-14:
I used ssh to open a terminal window on the BeagleBone Black and logged on as user debian. nano was used to edit files and the programs were started directly from the command line. Debian 9.2 iot was used.

I just tried the way you did with cloud9. Everything works as expected and this is what can be seen in the cloud9 console when the "input" test program is run:
```
Debugger listening on [::]:15454
4:4:91-ti-r133
0

Exiting Cleanly
Unable to open /dev/mem
Unable to open /dev/mem
ERROR: trying to disable motors before they have been initialized
Unable to open /dev/mem
Unable to open /dev/mem
Unable to open /dev/mem
```
I saw the same when running the program in a terminal logged on as user debian.

fivdi, 2017-10-14:
I'm afraid I don't know why your system is behaving differently.

udoom, 2017-10-14:
very strange. 
did you test it on a debian 9.2 iot "just installed" without any other config ?

I understand the js script error because : 

```
# ls -la /sys/devices/platform/ocp

root@beaglebone:~# ls -la /sys/devices/platform/ocp
total 0
drwxr-xr-x 39 root root    0 Oct 14 21:20 .
drwxr-xr-x 21 root root    0 Jan  1  2000 ..
drwxr-xr-x  3 root root    0 Jan  1  2000 40300000.ocmcram
drwxr-xr-x  3 root root    0 Jan  1  2000 40302000.ocmcram_nocache
drwxrwxr-x  4 root gpio    0 Jan  1  2000 44e07000.gpio
drwxr-xr-x  4 root root    0 Jan  1  2000 44e09000.serial
drwxr-xr-x  4 root root    0 Jan  1  2000 44e0b000.i2c
drwxr-xr-x  5 root root    0 Jan  1  2000 44e35000.wdt
drwxr-xr-x  4 root root    0 Jan  1  2000 44e3e000.rtc
drwxr-xr-x  9 root root    0 Jan  1  2000 47400000.usb
drwxr-xr-x  3 root root    0 Jan  1  2000 48038000.mcasp
drwxr-xr-x  3 root root    0 Jan  1  2000 48042000.timer
drwxr-xr-x  3 root root    0 Jan  1  2000 48044000.timer
drwxr-xr-x  3 root root    0 Jan  1  2000 48046000.timer
drwxr-xr-x  3 root root    0 Jan  1  2000 48048000.timer
drwxr-xr-x  3 root root    0 Jan  1  2000 4804a000.timer
drwxrwxr-x  4 root gpio    0 Jan  1  2000 4804c000.gpio
drwxr-xr-x  4 root root    0 Jan  1  2000 48060000.mmc
drwxr-xr-x  4 root root    0 Jan  1  2000 480c8000.mailbox
drwxr-xr-x  3 root root    0 Jan  1  2000 480ca000.spinlock
drwxr-xr-x  4 root root    0 Jan  1  2000 4819c000.i2c
drwxrwxr-x  4 root gpio    0 Jan  1  2000 481ac000.gpio
drwxrwxr-x  4 root gpio    0 Jan  1  2000 481ae000.gpio
drwxr-xr-x  4 root root    0 Jan  1  2000 481d8000.mmc
drwxr-xr-x  3 root root    0 Jan  1  2000 48200000.interrupt-controller
drwxr-xr-x  5 root root    0 Jan  1  2000 4830e000.lcdc
drwxr-xr-x  3 root root    0 Jan  1  2000 48310000.rng
drwxr-xr-x  4 root root    0 Jan  1  2000 49000000.edma
drwxr-xr-x  3 root root    0 Jan  1  2000 49800000.tptc
drwxr-xr-x  3 root root    0 Jan  1  2000 49900000.tptc
drwxr-xr-x  3 root root    0 Jan  1  2000 49a00000.tptc
drwxr-xr-x  7 root root    0 Jan  1  2000 4a100000.ethernet
drwxr-xr-x  6 root root    0 Jan  1  2000 4a300000.pruss
drwxr-xr-x  3 root root    0 Jan  1  2000 4c000000.emif
drwxr-xr-x  3 root root    0 Jan  1  2000 53100000.sham
drwxr-xr-x  3 root root    0 Jan  1  2000 53500000.aes
drwxr-xr-x  3 root root    0 Jan  1  2000 56000000.sgx
-rw-r--r--  1 root root 4096 Oct 14 21:20 driver_override
-r--r--r--  1 root root 4096 Oct 14 21:20 modalias
drwxr-xr-x  6 root root    0 Jan  1  2000 ocp:l4_wkup@44c00000
lrwxrwxrwx  1 root root    0 Oct 14 21:20 of_node -> ../../../firmware/devicetree/base/ocp
drwxr-xr-x  2 root root    0 Oct 14 21:20 power
lrwxrwxrwx  1 root root    0 Jan  1  2000 subsystem -> ../../../bus/platform
-rw-r--r--  1 root root 4096 Jan  1  2000 uevent
```

there is no "ocp:P9_41_pinmux" ...

how can I see if the universal cape is loaded with this new uboot tool ?

fivdi, 2017-10-14:
> did you test it on a debian 9.2 iot "just installed" without any other config ?

Yes, with one exception. The following lines were added to `/boot/uEnv.txt`:
```
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
```

> how can I see if the universal cape is loaded with this new uboot tool ?

I don't think it is being loaded, at lease not completely.

Does the command `dmesg` display and error messages that could help explain things?

Is there any interesting information in the files at `/var/log`?

udoom, 2017-10-14:
here is my dmesg : 
(I had a look on it, but I did not see anything interesting, maybe you will ?)

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.91-ti-r133 (root@b7-am57xx-beagle-x15-2gb) (gcc version 6.3.0 20170516 (Debian 6.3.0-18) ) #1 SMP Tue Oct 10 05:18:08 UTC 2017
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: TI AM335x BeagleBone Black
[    0.000000] cma: Reserved 48 MiB at 0x9c800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 130560
[    0.000000] free_area_init_node: node 0, pgdat c10d8f80, node_mem_map df961000
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon )
[    0.000000] PERCPU: Embedded 13 pages/cpu @df920000 s24268 r8192 d20788 u53248
[    0.000000] pcpu-alloc: s24268 r8192 d20788 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 444144K/522240K available (11720K kernel code, 948K rwdata, 3904K rodata, 756K init, 854K bss, 28944K reserved, 49152K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
                   lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0xc0008000 - 0xc0f4a584   (15626 kB)
                     .init : 0xc0f4b000 - 0xc1008000   ( 756 kB)
                     .data : 0xc1008000 - 0xc10f5108   ( 949 kB)
                      .bss : 0xc10f8000 - 0xc11cd820   ( 855 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=2 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=32, nr_cpu_ids=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000014] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000035] clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000048] OMAP clocksource: timer1 at 24000000 Hz
[    0.000842] clocksource_probe: no matching clocksources found
[    0.001049] Console: colour dummy device 80x30
[    0.001077] WARNING: Your 'console=ttyO0' has been replaced by 'ttyS0'
[    0.001085] This ensures that you still see kernel messages. Please
[    0.001091] update your kernel commandline.
[    0.001112] Calibrating delay loop... 995.32 BogoMIPS (lpj=1990656)
[    0.046813] pid_max: default: 32768 minimum: 301
[    0.046959] Security Framework initialized
[    0.046974] Yama: becoming mindful.
[    0.047012] AppArmor: AppArmor disabled by boot time parameter
[    0.047185] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.047199] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.048067] Initializing cgroup subsys io
[    0.048098] Initializing cgroup subsys memory
[    0.048146] Initializing cgroup subsys devices
[    0.048164] Initializing cgroup subsys freezer
[    0.048181] Initializing cgroup subsys net_cls
[    0.048195] Initializing cgroup subsys perf_event
[... 304 lines cut ...]
[   22.951821] tda998x 0-0070: found TDA19988
[   22.958451] tilcdc 4830e000.lcdc: bound 0-0070 (ops tda998x_ops [tda998x])
[   22.958489] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   22.958497] [drm] No driver support for vblank timestamp query.
[   22.965429] tilcdc 4830e000.lcdc: No connectors reported connected with modes
[   22.965472] [drm] Cannot find any crtc or sizes - going 1024x768
[   23.106603] Console: switching to colour frame buffer device 128x48
[   23.273413] tilcdc 4830e000.lcdc: fb0:  frame buffer device
[   23.273467] [drm] Initialized tilcdc 1.0.0 20121205 on minor 0
[   27.778054] asoc-simple-card sound: i2s-hifi <-> 48038000.mcasp mapping ok
```

in daemon.log some red words : 

```
Oct 14 21:45:09 beaglebone avahi-daemon[392]: chroot.c: open() failed: No such file or directory
Oct 14 21:45:09 beaglebone systemd[1]: Starting BB WL18xx wlan0 Service...
Oct 14 21:45:09 beaglebone avahi-daemon[374]: Failed to open /etc/resolv.conf: Invalid argument
[...]
Oct 14 21:45:10 beaglebone connmand[358]: Failed to open RFKILL control device
[...]
Oct 14 21:47:50 beaglebone systemd-udevd[293]: Process '/usr/sbin/alsactl -E HOME=/run/alsa restore 0' failed with exit code 99.
```

same errors in syslog

fivdi, 2017-10-15:
What I find interesting about the output of dmesg is what's not there rather than what's there.

On your system the output of dmesg contains the following two lines directly beside each other:
```
[    2.083555] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    2.086098] wkup_m3_ipc 44e11324.wkup_m3_ipc: could not get rproc handle
```

On my system there lots of additional output related to cape-universal between those two lines:
```
[    2.156163] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    2.157162] gpio-of-helper ocp:cape-universal: Allocated GPIO id=0
[    2.157472] gpio-of-helper ocp:cape-universal: Allocated GPIO id=1
[    2.157773] gpio-of-helper ocp:cape-universal: Allocated GPIO id=2
[    2.157932] gpio-of-helper ocp:cape-universal: Allocated GPIO id=3
[    2.158073] gpio-of-helper ocp:cape-universal: Allocated GPIO id=4
[    2.158219] gpio-of-helper ocp:cape-universal: Allocated GPIO id=5
[    2.158367] gpio-of-helper ocp:cape-universal: Allocated GPIO id=6
[    2.158504] gpio-of-helper ocp:cape-universal: Allocated GPIO id=7
[    2.158649] gpio-of-helper ocp:cape-universal: Allocated GPIO id=8
[    2.158795] gpio-of-helper ocp:cape-universal: Allocated GPIO id=9
[    2.158935] gpio-of-helper ocp:cape-universal: Allocated GPIO id=10
[    2.159077] gpio-of-helper ocp:cape-universal: Allocated GPIO id=11
[    2.159275] gpio-of-helper ocp:cape-universal: Allocated GPIO id=12
[    2.159421] gpio-of-helper ocp:cape-universal: Allocated GPIO id=13
[    2.159571] gpio-of-helper ocp:cape-universal: Allocated GPIO id=14
[    2.159713] gpio-of-helper ocp:cape-universal: Allocated GPIO id=15
[    2.159879] gpio-of-helper ocp:cape-universal: Allocated GPIO id=16
[    2.160032] gpio-of-helper ocp:cape-universal: Allocated GPIO id=17
[    2.160175] gpio-of-helper ocp:cape-universal: Allocated GPIO id=18
[    2.160316] gpio-of-helper ocp:cape-universal: Allocated GPIO id=19
[    2.160462] gpio-of-helper ocp:cape-universal: Allocated GPIO id=20
[    2.160603] gpio-of-helper ocp:cape-universal: Allocated GPIO id=21
[    2.160751] gpio-of-helper ocp:cape-universal: Allocated GPIO id=22
[    2.160931] gpio-of-helper ocp:cape-universal: Allocated GPIO id=23
[    2.161083] gpio-of-helper ocp:cape-universal: Allocated GPIO id=24
[    2.161234] gpio-of-helper ocp:cape-universal: Allocated GPIO id=25
[    2.161699] gpio-of-helper ocp:cape-universal: Allocated GPIO id=26
[    2.161875] gpio-of-helper ocp:cape-universal: Allocated GPIO id=27
[    2.162021] gpio-of-helper ocp:cape-universal: Allocated GPIO id=28
[    2.162155] gpio-of-helper ocp:cape-universal: Allocated GPIO id=29
[    2.162293] gpio-of-helper ocp:cape-universal: Allocated GPIO id=30
[    2.162431] gpio-of-helper ocp:cape-universal: Allocated GPIO id=31
[    2.162571] gpio-of-helper ocp:cape-universal: Allocated GPIO id=32
[    2.162710] gpio-of-helper ocp:cape-universal: Allocated GPIO id=33
[    2.162845] gpio-of-helper ocp:cape-universal: Allocated GPIO id=34
[    2.162978] gpio-of-helper ocp:cape-universal: Allocated GPIO id=35
[    2.163119] gpio-of-helper ocp:cape-universal: Allocated GPIO id=36
[    2.163256] gpio-of-helper ocp:cape-universal: Allocated GPIO id=37
[    2.163397] gpio-of-helper ocp:cape-universal: Allocated GPIO id=38
[    2.163532] gpio-of-helper ocp:cape-universal: Allocated GPIO id=39
[    2.163671] gpio-of-helper ocp:cape-universal: Allocated GPIO id=40
[    2.163811] gpio-of-helper ocp:cape-universal: Allocated GPIO id=41
[    2.163951] gpio-of-helper ocp:cape-universal: Allocated GPIO id=42
[    2.164091] gpio-of-helper ocp:cape-universal: Allocated GPIO id=43
[    2.164239] gpio-of-helper ocp:cape-universal: Allocated GPIO id=44
[    2.164375] gpio-of-helper ocp:cape-universal: Allocated GPIO id=45
[    2.164524] gpio-of-helper ocp:cape-universal: Allocated GPIO id=46
[    2.164661] gpio-of-helper ocp:cape-universal: Allocated GPIO id=47
[    2.164799] gpio-of-helper ocp:cape-universal: Allocated GPIO id=48
[    2.164940] gpio-of-helper ocp:cape-universal: Allocated GPIO id=49
[    2.165079] gpio-of-helper ocp:cape-universal: Allocated GPIO id=50
[    2.165219] gpio-of-helper ocp:cape-universal: Allocated GPIO id=51
[    2.165464] gpio-of-helper ocp:cape-universal: Allocated GPIO id=52
[    2.165810] gpio-of-helper ocp:cape-universal: Allocated GPIO id=53
[    2.165968] gpio-of-helper ocp:cape-universal: Allocated GPIO id=54
[    2.166111] gpio-of-helper ocp:cape-universal: Allocated GPIO id=55
[    2.166255] gpio-of-helper ocp:cape-universal: Allocated GPIO id=56
[    2.166401] gpio-of-helper ocp:cape-universal: Allocated GPIO id=57
[    2.166539] gpio-of-helper ocp:cape-universal: Allocated GPIO id=58
[    2.166681] gpio-of-helper ocp:cape-universal: Allocated GPIO id=59
[    2.166827] gpio-of-helper ocp:cape-universal: Allocated GPIO id=60
[    2.166967] gpio-of-helper ocp:cape-universal: Allocated GPIO id=61
[    2.167109] gpio-of-helper ocp:cape-universal: Allocated GPIO id=62
[    2.167256] gpio-of-helper ocp:cape-universal: Allocated GPIO id=63
[    2.167401] gpio-of-helper ocp:cape-universal: Allocated GPIO id=64
[    2.167547] gpio-of-helper ocp:cape-universal: Allocated GPIO id=65
[    2.167688] gpio-of-helper ocp:cape-universal: Allocated GPIO id=66
[    2.167698] gpio-of-helper ocp:cape-universal: ready
[    2.177603] wkup_m3_ipc 44e11324.wkup_m3_ipc: could not get rproc handle
```

Maybe there's something wrong with the boot loader. Does it help if the boot button is pressed and held pressed before and during power on?

Maybe there's information of interest sent to the [serial port](http://mysticengineering.com/debug.buddy/images/beaglebone-black-serial.jpg) while booting?

udoom, 2017-10-15:
you are right : 
keep "boot button" pressed fix the issue : I see now the gpio-of-helper in dmesg

and I understand why it did not work : 
I read this [boot-modes](https://github.com/beagleboard/beaglebone-black/wiki/System-Reference-Manual#535-boot-modes).
boot sequence is : 
1. eMMC
2. sd card
[...]
and on my eMMC, I had a old distribution where uboot was not activated.
that's why we did not see "gpio-of-helper ocp:cape-universal" in dmesg.

I flash my eMMC, and patch /boot/uEnv.txt, and it is working as expected.

thank's a lot for your help !

I close the issue

closed 2017-10-15 by udoom