jadonk/bonescript #156: PWM: changes for udev permissions

RobertCNelson, 2017-09-27:
See: https://github.com/RobertCNelson/omap-image-builder/issues/109

RobertCNelson, 2017-09-29:
Here's the details: CC @pdp7

pwmX -> pwm-X:Y

in v4.4.x: (pwmchip1)

```
pwm0
pwm1
```

in v4.11.x+ (pwmchip1)

```
pwm-1:0
pwm-1:1
```

pwm0 -> pwm-x:
pwm1 -> 

In v4.4.x
```
root@beaglebone:/sys/class/pwm# ls -lha
total 0
drwxrwxr-x  2 root pwm  0 Sep 29 20:57 .
drwxr-xr-x 56 root root 0 Sep 29 20:57 ..
lrwxrwxrwx  1 root pwm  0 Sep 29 20:57 pwmchip0 -> ../../devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip0
lrwxrwxrwx  1 root pwm  0 Sep 29 20:57 pwmchip2 -> ../../devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip2
lrwxrwxrwx  1 root pwm  0 Sep 29 20:57 pwmchip4 -> ../../devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip4
lrwxrwxrwx  1 root pwm  0 Sep 29 20:57 pwmchip6 -> ../../devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip6
lrwxrwxrwx  1 root pwm  0 Sep 29 20:57 pwmchip7 -> ../../devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip7
```
```
root@beaglebone:~# ls -lha /sys/class/pwm/pwmchip*/
/sys/class/pwm/pwmchip0/:
total 0
drwxrwxr-x 5 root pwm     0 Sep 29 20:58 .
drwxr-xr-x 3 root root    0 Sep 29 20:57 ..
lrwxrwxrwx 1 root pwm     0 Sep 29 20:57 device -> ../../../48300200.pwm
-rw-rw---- 1 root pwm  4.0K Sep 29 20:58 export
-rw-rw-r-- 1 root pwm  4.0K Sep 29 20:57 npwm
drwxrwxr-x 2 root pwm     0 Sep 29 20:57 power
drwxr-xr-x 3 root root    0 Sep 29 20:59 pwm0
drwxr-xr-x 3 root root    0 Sep 29 20:59 pwm1
lrwxrwxrwx 1 root pwm     0 Sep 29 20:57 subsystem -> ../../../../../../../class/pwm
-rw-rw-r-- 1 root pwm  4.0K Sep 29 20:57 uevent
-rw-rw---- 1 root pwm  4.0K Sep 29 20:57 unexport
```
in v4.11.x+
```
root@beaglebone:~# ls -lha /sys/class/pwm/
total 0
drwxrwxr-x  2 root pwm  0 Sep 29 21:09 .
drwxr-xr-x 61 root root 0 Sep 29 21:10 ..
lrwxrwxrwx  1 root pwm  0 Sep 29 21:10 pwmchip0 -> ../../devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0
lrwxrwxrwx  1 root pwm  0 Sep 29 21:10 pwmchip1 -> ../../devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1
lrwxrwxrwx  1 root pwm  0 Sep 29 21:10 pwmchip3 -> ../../devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3
lrwxrwxrwx  1 root pwm  0 Sep 29 21:10 pwmchip5 -> ../../devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip5
lrwxrwxrwx  1 root pwm  0 Sep 29 21:10 pwmchip6 -> ../../devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6
```
```
/sys/class/pwm/pwmchip1/:
total 0
drwxrwxr-x 5 root pwm     0 Sep 29 21:10 .
drwxr-xr-x 3 root root    0 Sep 29 21:09 ..
lrwxrwxrwx 1 root pwm     0 Sep 29 21:10 device -> ../../../48300200.pwm
-rw-rw---- 1 root pwm  4.0K Sep 29 21:10 export
-rw-rw-r-- 1 root pwm  4.0K Sep 29 21:09 npwm
drwxrwxr-x 2 root pwm     0 Sep 29 21:09 power
drwxrwxr-x 3 root pwm     0 Sep 29 21:10 pwm-1:0
drwxrwxr-x 3 root pwm     0 Sep 29 21:10 pwm-1:1
lrwxrwxrwx 1 root pwm     0 Sep 29 21:10 subsystem -> ../../../../../../../class/pwm
-rw-rw-r-- 1 root pwm  4.0K Sep 29 21:09 uevent
-rw-rw---- 1 root pwm  4.0K Sep 29 21:09 unexport
```

pdp7, 2017-09-30:
thanks @RobertCNelson

jadonk, 2017-12-20:
@RobertCNelson as far as I can tell, I'm already accessing the correct 4.4 file and the code should find the right locations on other kernels as you've described. Note that BoneScript uses the addresses, not the indexes for the search. This makes it more generic.

To test, I got super lazy and updated the udev rules to update all permissions any time *any* of the pwm drivers is loaded. There are so many events that the file was starting to be a pain otherwise. The ecap files will also need updates, but I didn't have them enabled on PocketBeagle to test.

I tried pointlessly to hack the udev rules to respond to export/unexport and I finally *really* know about https://github.com/beagleboard/linux/issues/124. 

Anyway, if you update the permissions, the head of BoneScript works.

```
debian@beaglebone:~/bonescript$ DEBUG=true node -pe "require('.').analogWrite('P2_1', 0.5)"
4:4:91-ti-r139
Using Mainline interface
debug: index.js loaded
debug: analogWrite(P2_1,0.5,);
debug: getPinMode(P2_1);
debug: pinMode(P2_1,analog_out,6,pulldown,fast);
debug: pinMode: pin.led = undefined
debug: hw.setPinMode(P2_1,6,bspwm,{"value":true});
debug: find_sysfsFile(ocp,/sys/devices,ocp.)
debug: is_ocp, found = undefined
debug: find_sysfsFile(ocp,/sys/devices/platform,ocp)
debug: is_ocp, found2 = /sys/devices/platform/ocp
debug: find_sysfsFile(ocp:P2_1_pinmux,/sys/devices/platform/ocp,ocp:P9_14_pinmux,ocp:P2_01_pinmux,ocp:PWM_pinmux)
debug: chipPath = /sys/devices/platform/ocp/48302000.epwmss
debug: addrPath = /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm
debug: pwmchipPath = /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip2
debug: pwmPath = /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip2/pwm0
fs.js:642
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                 ^

Error: EACCES: permission denied, open '/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip2/pwm0/enable'
  at Error (native)
  at Object.fs.openSync (fs.js:642:18)
  at Object.fs.writeFileSync (fs.js:1348:33)
  at Object.fs.appendFileSync (fs.js:1407:6)
  at Object.exports.setPinMode (/home/debian/bonescript/src/hw_mainline.js:125:12)
  at Object.f.pinMode (/home/debian/bonescript/src/index.js:179:15)
  at Object.f.analogWrite (/home/debian/bonescript/src/index.js:455:11)
  at [eval]:1:14
  at ContextifyScript.Script.runInThisContext (vm.js:25:33)
  at Object.runInThisContext (vm.js:97:38)
  at Object.<anonymous> ([eval]-wrapper:6:22)
  at Module._compile (module.js:570:32)
  at evalScript (bootstrap_node.js:347:27)
  at run (bootstrap_node.js:122:11)
  at run (bootstrap_node.js:383:7)
  at startup (bootstrap_node.js:121:9)
  at bootstrap_node.js:496:3

debian@beaglebone:~/bonescript$ sudo chown -R root:pwm /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip2/pwm0
debian@beaglebone:~/bonescript$ sudo chmod -R ug+rw /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip2/pwm0
debian@beaglebone:~/bonescript$ DEBUG=true node -pe "require('.').analogWrite('P2_1', 0.5)"4:4:91-ti-r139
Using Mainline interface
debug: index.js loaded
debug: analogWrite(P2_1,0.5,);
debug: getPinMode(P2_1);
debug: pinMode(P2_1,analog_out,6,pulldown,fast);
debug: pinMode: pin.led = undefined
debug: hw.setPinMode(P2_1,6,bspwm,{"value":true});
debug: find_sysfsFile(ocp,/sys/devices,ocp.)
debug: is_ocp, found = undefined
debug: find_sysfsFile(ocp,/sys/devices/platform,ocp)
debug: is_ocp, found2 = /sys/devices/platform/ocp
debug: find_sysfsFile(ocp:P2_1_pinmux,/sys/devices/platform/ocp,ocp:P9_14_pinmux,ocp:P2_01_pinmux,ocp:PWM_pinmux)
debug: chipPath = /sys/devices/platform/ocp/48302000.epwmss
debug: addrPath = /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm
debug: pwmchipPath = /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip2
debug: pwmPath = /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip2/pwm0
debug: hw.writePWMFreqAndValue(P2_1,{ key: 'P2_1', freq: 0 },2000,0.5,[object Object]);
debug: hw.writePWMFreqAndValue: pwm.freq=0, freq=2000, period=500000
debug: Updating PWM period: 500000
debug: Updating PWM duty: 250000
true
debug: Calling run()

Exiting Cleanly
Unable to open /dev/mem
Unable to open /dev/mem
ERROR: trying to disable motors before they have been initialized
Unable to open /dev/mem
Unable to open /dev/mem
Unable to open /dev/mem
debian@beaglebone:~/bonescript$
```

closed 2017-12-20 by jadonk

RobertCNelson, 2017-12-21:
@jadonk we are closer, just rebuild bonescript for buster: (bonescript-0.6.2-beta3-08f082b-v8.9.3)

https://rcn-ee.net/rootfs/bb.org/testing/2017-12-21/buster-iot/

There still an issue with v4.14.x (root and debian), udev permissions are correct for "debian", but considering root is failing, something else causing an issue.

v4.4.x: (as root = works)
```
root@beaglebone:~# DEBUG=true node -pe "require('bonescript').analogWrite('P8_13', 0.5)"
4:4:91-ti-r140
Using Mainline interface
debug: index.js loaded
debug: analogWrite(P8_13,0.5,);
debug: getPinMode(P8_13);
debug: pinMode(P8_13,analog_out,4,pulldown,fast);
debug: pinMode: pin.led = undefined
debug: hw.setPinMode(P8_13,4,bspwm,{"value":true});
debug: find_sysfsFile(ocp,/sys/devices,ocp.)
debug: is_ocp, found = undefined
debug: find_sysfsFile(ocp,/sys/devices/platform,ocp)
debug: is_ocp, found2 = /sys/devices/platform/ocp
debug: find_sysfsFile(ocp:P8_13_pinmux,/sys/devices/platform/ocp,ocp:P8_13_pinmux,ocp:T10_pinmux)
debug: chipPath = /sys/devices/platform/ocp/48304000.epwmss
debug: addrPath = /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm
debug: pwmchipPath = /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6
debug: pwmPath = undefined
debug: pwmPath = /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm1
debug: hw.writePWMFreqAndValue(P8_13,{ key: 'P8_13', freq: 0 },2000,0.5,[object Object]);
debug: hw.writePWMFreqAndValue: pwm.freq=0, freq=2000, period=500000
debug: Updating PWM period: 500000
debug: Updating PWM duty: 250000
true
debug: Calling run()

Exiting Cleanly
ERROR: trying to disable motors before they have been initialized
```
v4.14.x (as root)

```
root@beaglebone:~$ DEBUG=true node -pe "require('bonescript').analogWrite('P8_13', 0.5)"
4:14:7-ti-r18
Using Mainline interface
debug: index.js loaded
debug: analogWrite(P8_13,0.5,);
debug: getPinMode(P8_13);
debug: getPinMode(P8_13): TypeError: Cannot read property '2' of null
debug: pinMode(P8_13,analog_out,4,disabled,fast);
debug: pinMode: pin.led = undefined
debug: hw.setPinMode(P8_13,12,bspwm,{"value":true});
debug: find_sysfsFile(ocp,/sys/devices,ocp.)
debug: is_ocp, found = undefined
debug: find_sysfsFile(ocp,/sys/devices/platform,ocp)
debug: is_ocp, found2 = /sys/devices/platform/ocp
debug: find_sysfsFile(ocp:P8_13_pinmux,/sys/devices/platform/ocp,ocp:P8_13_pinmux,ocp:T10_pinmux)
debug: chipPath = /sys/devices/platform/ocp/48304000.epwmss
debug: addrPath = /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm
debug: pwmchipPath = /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6
debug: pwmPath = /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm-6:1
fs.js:735
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EINVAL: invalid argument, write
  at Object.fs.writeSync (fs.js:735:20)
  at Object.fs.writeFileSync (fs.js:1301:24)
  at Object.fs.appendFileSync (fs.js:1337:6)
  at Object.exports.setPinMode (/usr/local/lib/node_modules/bonescript/src/hw_mainline.js:133:12)
  at Object.f.pinMode (/usr/local/lib/node_modules/bonescript/src/index.js:179:15)
  at Object.f.analogWrite (/usr/local/lib/node_modules/bonescript/src/index.js:455:11)
[... 44 lines cut ...]
  at ContextifyScript.Script.runInThisContext (vm.js:50:33)
  at Object.runInThisContext (vm.js:139:38)
  at Object.<anonymous> ([eval]-wrapper:6:22)
  at Module._compile (module.js:635:30)
  at evalScript (bootstrap_node.js:462:27)
  at startup (bootstrap_node.js:163:9)
  at bootstrap_node.js:608:3
```
v4.14.x udev:
```
ls -lha /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm-6:1/
total 0
drwxrwxr-x 3 root pwm    0 Dec 21 20:20 .
drwxrwxr-x 4 root pwm    0 Dec 21 20:20 ..
-rw-rw-r-- 1 root pwm 4.0K Dec 21 20:20 capture
lrwxrwxrwx 1 root pwm    0 Dec 21 20:20 device -> ../../pwmchip6
-rw-rw-r-- 1 root pwm 4.0K Dec 21 20:20 duty_cycle
-rw-rw-r-- 1 root pwm 4.0K Dec 21 20:20 enable
-rw-rw-r-- 1 root pwm 4.0K Dec 21 20:20 period
-rw-rw-r-- 1 root pwm 4.0K Dec 21 20:20 polarity
drwxrwxr-x 2 root pwm    0 Dec 21 20:20 power
lrwxrwxrwx 1 root pwm    0 Dec 21 20:20 subsystem -> ../../../../../../../../class/pwm
-rw-rw-r-- 1 root pwm 4.0K Dec 21 20:20 uevent
```
Regards,