jadonk/bonescript #158: AnalogWrite error

el-j, 2017-12-13:
i use the pocketbeagle board.

and set pin p2_1 to analogwrite
```
b.pinMode(pin, b.ANALOG_OUTPUT, );
b.analogWrite(pin, 1, 2000, printJSON)
```

sadly it ends in this error:
```
{"err":"error updating PWM freq and value: undefined, Error: ENOENT: no such file or directory, open 'undefined/period'"}
```

same error in the playground. 
i really need analog write. do you have any suggestions how i can use it?

jadonk, 2017-12-20:
This is going to require a kernel update as I've got the other fixes in. I believe @RobertCNelson will do the kernel update soon. See https://github.com/jadonk/bonescript/issues/156.

jadonk, 2018-02-03:
Almost there. Still have the udev issue https://github.com/jadonk/bonescript/issues/161.

closed 2018-04-25 by jadonk