jadonk/bonescript #160: analogRead (and other) documentation

troywweber7, 2017-12-29:
This is just a minor feature request for better api documentation (either in a README or as a JSDoc for each function).

As an example, regarding `analogRead`, I noticed that it [divides a reading by 1800](https://github.com/jadonk/bonescript/blob/08f082b5458e00a424a1d9f8b70efa2c6190f45d/src/hw_universal.js#L204) in the code and I'm confused about why because I thought the ADC's were 12-bit which would lead me to believe one should be dividing a reading by 4096 (2^12).

Perhaps I am wrong about that last part, but I am still left thinking "When I do an `bonescript.analogRead('P9_39')` how do I map that value? Am I assuming correctly that the `analogRead` function returns a value of 0 ≤ VALUE < 1?"

fivdi, 2018-01-10:
The code that's used by `analogRead` today is [here](https://github.com/jadonk/bonescript/blob/master/src/hw_mainline.js#L232-L259). It uses 4095 (2^12 -1.)

The code you link to above is from a long time ago. Back then the raw analog values returned by the driver were value between 0 and 1800 (inclusive.) It's a millivolt value. The maximum value that can be input on analog input pins is 1.8V which is 1800 millivolts.

closed 2018-02-03 by jadonk

troywweber7, 2018-04-27:
Thanks for pointed to the proper code. I think it still might be a good idea (in a README for example) to document each of these functions unless they going to change drastically in the near future.