jadonk/bonescript #161: PWM: v4.14.x: (udev delays)

RobertCNelson, 2018-01-02:
Fix in Adafruit-lib: 100ms delay:

https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/c_pwm.c#L415-L418

On first startup, the pwm will fail in user mode:

```
debian@beaglebone:~$ DEBUG=true node -pe "require('bonescript').analogWrite('P8_13', 0.5)"
4:14:10-ti-r21
Using Mainline interface
debug: index.js loaded
debug: analogWrite(P8_13,0.5,);
debug: getPinMode(P8_13);
debug: getPinMode(P8_13): TypeError: Cannot read property '2' of null
debug: pinMode(P8_13,analog_out,4,disabled,fast);
debug: pinMode: pin.led = undefined
debug: hw.setPinMode(P8_13,12,bspwm,{"value":true});
debug: find_sysfsFile(ocp,/sys/devices,ocp.)
debug: is_ocp, found = undefined
debug: find_sysfsFile(ocp,/sys/devices/platform,ocp)
debug: is_ocp, found2 = /sys/devices/platform/ocp
debug: find_sysfsFile(ocp:P8_13_pinmux,/sys/devices/platform/ocp,ocp:P8_13_pinmux,ocp:T10_pinmux)
debug: chipPath = /sys/devices/platform/ocp/48304000.epwmss
debug: addrPath = /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm
debug: pwmchipPath = /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6
debug: pwmPath = undefined
debug: pwmPath = /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm-6:1
debug: hw.writePWMFreqAndValue(P8_13,{ key: 'P8_13', freq: 0 },2000,0.5,[object Object]);
debug: hw.writePWMFreqAndValue: pwm.freq=0, freq=2000, period=500000
debug: Stopping PWM
debug: Error stopping PWM: Error: EACCES: permission denied, open '/sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm-6:1/enable'
debug: Updating PWM period: 500000
info: Unable to update PWM period, period is set to 0
	Is other half of PWM enabled?
debug: Starting PWM
debug: Error starting PWM: Error: EACCES: permission denied, open '/sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm-6:1/enable'
debug: Updating PWM duty: 0
error: error updating PWM freq and value: /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm-6:1, Error: EACCES: permission denied, open '/sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm-6:1/duty_cycle'
true
debug: Calling run()

Exiting Cleanly
Unable to open /dev/mem
Unable to open /dev/mem
ERROR: trying to disable motors before they have been initialized
Unable to open /dev/mem
Unable to open /dev/mem
Unable to open /dev/mem
```
However, If you re-run it, it works fine, as udev has had time to fix the permissions..

```
debian@beaglebone:~$ DEBUG=true node -pe "require('bonescript').analogWrite('P8_13', 0.5)"
4:14:10-ti-r21
Using Mainline interface
debug: index.js loaded
debug: analogWrite(P8_13,0.5,);
debug: getPinMode(P8_13);
debug: getPinMode(P8_13): TypeError: Cannot read property '2' of null
debug: pinMode(P8_13,analog_out,4,disabled,fast);
debug: pinMode: pin.led = undefined
debug: hw.setPinMode(P8_13,12,bspwm,{"value":true});
debug: find_sysfsFile(ocp,/sys/devices,ocp.)
debug: is_ocp, found = undefined
debug: find_sysfsFile(ocp,/sys/devices/platform,ocp)
debug: is_ocp, found2 = /sys/devices/platform/ocp
debug: find_sysfsFile(ocp:P8_13_pinmux,/sys/devices/platform/ocp,ocp:P8_13_pinmux,ocp:T10_pinmux)
debug: chipPath = /sys/devices/platform/ocp/48304000.epwmss
debug: addrPath = /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm
debug: pwmchipPath = /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6
debug: pwmPath = /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm-6:1
debug: hw.writePWMFreqAndValue(P8_13,{ key: 'P8_13', freq: 0 },2000,0.5,[object Object]);
debug: hw.writePWMFreqAndValue: pwm.freq=0, freq=2000, period=500000
debug: Stopping PWM
debug: Error stopping PWM: Error: EINVAL: invalid argument, write
debug: Updating PWM period: 500000
debug: Starting PWM
debug: Updating PWM duty: 250000
true
debug: Calling run()

Exiting Cleanly
Unable to open /dev/mem
Unable to open /dev/mem
ERROR: trying to disable motors before they have been initialized
Unable to open /dev/mem
Unable to open /dev/mem
Unable to open /dev/mem
```

jadonk, 2018-01-04:
The JavaScript library should be asynchronous. What system event can I use to trigger the update rather than an arbitrary stall that could be unreliable?

RobertCNelson, 2018-01-04:
can you loop over the dir? it should change from root:root -> root:pwm

RobertCNelson, 2018-01-04:
This would be overkill:

https://www.npmjs.com/package/udev

Regards,

fivdi, 2018-01-23:
The approach taken by beaglebone-io is to repeatedly access the files until there's no exception.

see https://github.com/julianduque/beaglebone-io/blob/v3.0.0/lib/pwm-output.js#L20-L53
and https://github.com/julianduque/beaglebone-io/blob/v3.0.0/lib/pwm-output.js#L98-L104

As mentioned in the comments it's a bit of a hack but it works.

closed 2018-04-25 by jadonk