jadonk/bonescript #182: pinMode fails on PocketBeagle

mzwtjp, 2018-10-29:
Hello.
I am using PocketBeagle.

> BeagleBoard.org Debian Image 2018-06-17
> bonescript@0.7.0 

I am trying gpio operation using BoneScript.
But, I get the following error when I try to set p1_2 gpio pin to output.

>$ node -pe "require('bonescript').pinMode('P1_2', 'output')"
>info: pinMode: Missing mux name for pin object: {"name":"AIN6","ain":6,"eeprom":73,"scale":4096,"key":"P1_2","ball":{"ZCZ":"A8","BSM":"C9"}}
>info: No pinmux for P1_2
>false
>info: running rc_cleanup()

I tried other names as 'gpio_87', etc. but they also did not work.

Whereas, I have no such problem with config-pin command.

$ config-pin p1_2 output
$ config-pin p1_2 hi
$ config-pin p1_2 lo
...

i.e.
In comparison, using config-pin command on PocketBeagle, I can set p1_2 pin ho/low properly).  But using bonescript I cannot.

mzwtjp, 2018-10-29:
As a try, I modified /usr/local/lib/node_modules/bonescript/src/bone.js as follows

>    {
>        "name": "GPIO2_23",
>        "gpio": 87,
>        "mux": "lcd_hsync",
>        "eeprom": 58,
>        "key": ["P8_29", "SERVO_3", "P1_2", "PB1_2"], // PB1_2 added.
>        "universalName": ["ocp:P1_02_pinmux"], // Added.
> ...

Then tried again with following.

>$ node -pe "require('bonescript').pinMode('PB1_2', 'out')"

Now I do not have error, but digitalWrite high/low still does not work.
(I did not look into further today.)

Is this just with P1.02 of PB (other pins should be fine)?

Anyway, the latest bonescript should be fine with the usage like below?
(This is one of the things I want to do..)

>var b = require('bonescript');
>b.pinMode("p1_2", b.OUTPUT);  // use P1.02 for high/low output
>b.digitalWrite("p1_2", b.HIGH);  // make it high
>b.digitalWrite("p1_2", b.LOW);   // make it low

mzwtjp, 2018-10-30:
Thanks it did work.

So I should be doing is to find corresponding BeagleBone pin number, and use it as the pin name to be passed to bonescript.
(P1.02 -> GPIO87 -> P8.29?)

What I am not sure is the numbers like GPIO87.  Is it AM335x signal name or OSD335x name?
If I were to look up myself, what I should be doing?

PB schematics:  P1.2 -> AN6/GPIO87
PB schematics: AN6/GPIO87 -> F2 (LCD_HSYNC) of SIP D
...?

fivdi, 2018-11-11:
>So I should be doing is to find corresponding BeagleBone pin number, and use it as the pin name to be passed to bonescript.
>(P1.02 -> GPIO87 -> P8.29?)

The bug that you detected only exists for P1_2 on the PocketBeagle. Unfortunately, P1_2 is defined twice in bone.js, it should only be defined once. For all other pins using Pn.m should function correctly on the PocketBeagle. The corresponding pin on the BeagleBone Black can be found the way you suggest.

>What I am not sure is the numbers like GPIO87. Is it AM335x signal name or OSD335x name?

The numbers like GPIO87 are required for Linux. They are Linux numbers. There are files in directory `/sys/class/gpio` on Linux systems that can be used to access GPIOs. The files in this directory require each GPIO to have a unique number.

>  If I were to look up myself, what I should be doing?

I use the images [here](http://beagleboard.org/Support/bone101) for the BeagleBone Black and the image [here](https://github.com/beagleboard/pocketbeagle/blob/master/docs/PocketBeagle_pinout.png) for the PocketBeagle.