jadonk/bonescript #189: BBAI: pin P9_15 output failure

pdp7, 2020-05-27:
[Reported in the Adafruit forum by jgomes78](https://forums.adafruit.com/viewtopic.php?f=49&t=165787) and posting here so that others can see the issue and eventual solution:
> I have a BeagleBone AI and I can't flash the LED on pin P9_15.
> I researched and verified that the mapping of the BB AI pins is different from the BBB.

Followup from jgomes78:
> I thank you for your attention, I did what you asked for, updated Debian, tried again to make the led blink on pin P9_15 and it didn't blink.
> I am also sending the command response you asked for and the LED flashing program. In this program USR0, USR1, USR2 and USR3 flash, but output P9_15 does not respond.

**blinkLED.js**
```javascript
#!/usr/bin/env node
////////////////////////////////////////
// blinkLED.js
// Blinks the USR LEDs and P9_15.
// Wiring: P9_15 connects to the plus lead of an LED. The negative lead of the
// LED goes to a 220 Ohm resistor. The other lead of the resistor goes
// to ground.
// Setup:
// See:
////////////////////////////////////////
const b = require('bonescript');

const leds = ["USR0", "USR1", "USR2", "USR3", "P9_15"];

for(var i in leds) {
b.pinMode(leds[i], b.OUTPUT);
}

var state = b.HIGH;
for(var i in leds) {
b.digitalWrite(leds[i], state);
}

setInterval(toggle, 100);

function toggle() {
if(state == b.LOW)
state = b.HIGH;
else
state = b.LOW;
for(var i in leds) {
b.digitalWrite(leds[i], state);
}
}
```

**version.sh**
```
git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
model:[BeagleBoard.org_BeagleBone_AI]
dogtag:[BeagleBoard.org Debian Buster IoT TIDL Image 2020-04-06]
UBOOT: Booted Device-Tree:[am5729-beagleboneai.dts]
kernel:[4.14.108-ti-r131]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200403.0-0rcnee0~buster+20200403]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~buster+20200322]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyS0,115200n8 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[ 3.642924] remoteproc remoteproc0: 4b234000.pru is available
[ 3.643683] remoteproc remoteproc1: 4b238000.pru is available
[ 3.658217] remoteproc remoteproc2: 4b2b4000.pru is available
[ 3.662158] remoteproc remoteproc3: 4b2b8000.pru is available
[ 19.166466] remoteproc remoteproc4: 58820000.ipu is available
[ 19.176372] remoteproc remoteproc5: 55020000.ipu is available
[ 19.189863] remoteproc remoteproc6: 40800000.dsp is available
[ 19.220370] remoteproc remoteproc7: 41000000.dsp is available
[ 19.329163] remoteproc remoteproc5: powering up 55020000.ipu
[ 19.329210] remoteproc remoteproc5: Booting fw image dra7-ipu2-fw.xem4, size 3751356
[ 19.466434] remoteproc remoteproc4: powering up 58820000.ipu
[ 19.466483] remoteproc remoteproc4: Booting fw image dra7-ipu1-fw.xem4, size 7051536
[ 19.839136] remoteproc remoteproc5: registered virtio0 (type 7)
[ 19.839159] remoteproc remoteproc5: remote processor 55020000.ipu is now up
[ 19.868918] remoteproc remoteproc7: powering up 41000000.dsp
[ 19.868967] remoteproc remoteproc7: Booting fw image dra7-dsp2-fw.xe66, size 21014532
[ 19.879184] remoteproc remoteproc4: registered virtio1 (type 7)
[ 19.879208] remoteproc remoteproc4: remote processor 58820000.ipu is now up
[ 19.942439] remoteproc remoteproc7: registered virtio2 (type 7)
[ 19.942463] remoteproc remoteproc7: remote processor 41000000.dsp is now up
[ 20.036706] Modules linked in: omap_remoteproc virtio_rpmsg_bus rpmsg_core cmemk(O) uio_pdrv_genirq uio spidev pruss_soc_bus pru_rproc pruss pruss_intc usbhid
[ 20.037822] Modules linked in: omap_remoteproc virtio_rpmsg_bus rpmsg_core cmemk(O) uio_pdrv_genirq uio spidev pruss_soc_bus pru_rproc pruss pruss_intc usbhid
[ 20.059156] remoteproc remoteproc6: powering up 40800000.dsp
[ 20.059223] remoteproc remoteproc6: Booting fw image dra7-dsp1-fw.xe66, size 21014532
[ 20.178648] remoteproc remoteproc6: registered virtio3 (type 7)
[ 20.178671] remoteproc remoteproc6: remote processor 40800000.dsp is now up
dmesg | grep pru
[ 3.632316] pruss 4b200000.pruss: creating PRU cores and other child platform devices
[ 3.642924] remoteproc remoteproc0: 4b234000.pru is available
[ 3.643005] pru-rproc 4b234000.pru: PRU rproc node /ocp/pruss_soc_bus@4b226004/pruss@0/pru@34000 probed successfully
[ 3.643683] remoteproc remoteproc1: 4b238000.pru is available
[ 3.643732] pru-rproc 4b238000.pru: PRU rproc node /ocp/pruss_soc_bus@4b226004/pruss@0/pru@38000 probed successfully
[ 3.647007] pruss 4b280000.pruss: creating PRU cores and other child platform devices
[ 3.658217] remoteproc remoteproc2: 4b2b4000.pru is available
[ 3.658277] pru-rproc 4b2b4000.pru: PRU rproc node /ocp/pruss_soc_bus@4b2a6004/pruss@0/pru@34000 probed successfully
[ 3.662158] remoteproc remoteproc3: 4b2b8000.pru is available
[ 3.662229] pru-rproc 4b2b8000.pru: PRU rproc node /ocp/pruss_soc_bus@4b2a6004/pruss@0/pru@38000 probed successfully
[ 20.036706] Modules linked in: omap_remoteproc virtio_rpmsg_bus rpmsg_core cmemk(O) uio_pdrv_genirq uio spidev pruss_soc_bus pru_rproc pruss pruss_intc usbhid
[ 20.037822] Modules linked in: omap_remoteproc virtio_rpmsg_bus rpmsg_core cmemk(O) uio_pdrv_genirq uio spidev pruss_soc_bus pru_rproc pruss pruss_intc usbhid
[ 34.410084] pruss_uio_shmem 4b200000.pruss_shmem: Allocating gdev
[ 34.410189] pruss_uio_shmem 4b200000.pruss_shmem: Allocating info
[ 34.410211] pruss_uio_shmem 4b200000.pruss_shmem: Requesting resource
[ 34.410376] pruss_uio_shmem 4b200000.pruss_shmem: Mapping resource
[ 34.411319] pruss_uio_shmem 4b200000.pruss_shmem: Registering with uio driver
[ 34.414375] pruss_uio_shmem 4b200000.pruss_shmem: Saving platform data
[ 34.415159] pruss_uio_shmem 4b280000.pruss_shmem: Allocating gdev
[ 34.415259] pruss_uio_shmem 4b280000.pruss_shmem: Allocating info
[ 34.415346] pruss_uio_shmem 4b280000.pruss_shmem: Requesting resource
[ 34.415404] pruss_uio_shmem 4b280000.pruss_shmem: Mapping resource
[ 34.415577] pruss_uio_shmem 4b280000.pruss_shmem: Registering with uio driver
[ 34.442375] pruss_uio_shmem 4b280000.pruss_shmem: Saving platform data
dmesg | grep pinctrl-single
[ 0.781755] pinctrl-single 4a003400.pinmux: 282 pins at pa fc003400 size 1128
dmesg | grep gpio-of-helper
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0c40:8000
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

pdp7, 2020-05-27:
I've asked the user to run:
`/opt/scripts/device/bone/show-pins.pl`

pdp7, 2020-05-28:
User has update the output of show-pins.pl and it does look like it is in gpio mode:
```
P9.19a 16 R6 7 fast rx up i2c4_scl
P9.20a 17 T9 7 fast rx up i2c4_sda
P8.35b 57 AD9 e fast down gpio3_0
P8.33b 58 AF9 e fast down gpio3_1
P9.21a 60 AF8 e fast down gpio3_3
P8.12 67 AG6 e fast down gpio3_10
P8.11 68 AH4 e fast down gpio3_11
P9.15 69 AG4 e fast down gpio3_12
P9.26b 81 AE2 e fast down gpio3_24
P8.34b 89 G6 e fast down gpio4_0
P8.36b 90 F2 e fast down gpio4_1
P8.15a 92 D1 e fast down gpio4_3
P9.20b 94 D2 c fast rx up pr1_pru1_gpi1
P9.19b 95 F4 c fast rx up pr1_pru1_gpi2
P9.41b 96 C1 e fast down gpio4_7
P8.18 98 F5 c fast rx pr1_pru1_gpi5
P8.19 99 E6 c fast rx pr1_pru1_gpi6
P8.13 100 D3 c fast rx pr1_pru1_gpi7
P8.14 102 D5 e fast down gpio4_13
P9.42b 103 C2 e fast down gpio4_14
P9.27a 104 C3 e fast down gpio4_15
P9.14 107 D6 a fast down ehrpwm3A
P9.16 108 C5 d fast down pr1_pru1_gpo15
P8.15b 109 A3 d fast down pr1_pru1_gpo16
P8.26 110 B3 d fast down pr1_pru1_gpo17
P8.16 111 B4 d fast down pr1_pru1_gpo18
P8.28a 114 D11 e fast down gpio4_19
P8.30a 115 B10 e fast down gpio4_20
P8.29a 117 C11 e fast down gpio4_22
P8.27a 118 E11 e fast down gpio4_23
P8.45a 119 F11 e fast down gpio8_0
P8.46a 120 G10 e fast down gpio8_1
P8.43 121 F10 e fast down gpio8_2
P8.44 122 G11 e fast down gpio8_3
P8.41 123 E9 e fast down gpio8_4
P8.42 124 F9 e fast down gpio8_5
P8.39 125 F8 e fast down gpio8_6
P8.40 126 E7 e fast down gpio8_7
P8.37a 127 E8 e fast down gpio8_8
P8.38a 128 D9 e fast down gpio8_9
P8.36a 129 D7 e fast down gpio8_10
P8.34a 130 D8 e fast down gpio8_11
P8.35a 131 A5 e fast down gpio8_12
P8.33a 132 C6 e fast down gpio8_13
P8.31a 133 C8 e fast down gpio8_14
P8.32a 134 C7 e fast down gpio8_15
P8.45b 135 B7 e fast down gpio8_16
P9.11b 136 B8 e fast down gpio8_17
P8.17 137 A7 e fast down gpio8_18
P8.27b 138 A8 e fast down gpio8_19
P8.28b 139 C9 e fast down gpio8_20
P8.29b 140 A9 e fast down gpio8_21
P8.30b 141 B9 e fast down gpio8_22
P8.46b 142 A10 e fast down gpio8_23
P9.13b 160 AB10 f fast down Driver off
P9.26a 162 E21 e fast down gpio6_14
P9.24 163 F20 e fast down gpio6_15
P9.25 165 D18 e fast down gpio6_17
P8.09 166 E17 e fast down gpio6_18
P9.22a 167 B26 e fast down gpio6_19
P9.41a 168 C23 e fast down gpio6_20
P9.31b 169 C14 e fast down gpio7_31
P9.29b 170 D14 e fast down gpio7_30
P9.12 171 B14 e fast down gpio5_0
P9.27b 172 J14 e fast down gpio5_1
P9.18b 173 G12 e fast down gpio5_2
P9.17b 174 F12 e fast down gpio5_3
P9.31a 181 B12 e fast down gpio5_10
P9.29a 182 A11 e fast down gpio5_11
P9.30 183 B13 e fast down gpio5_12
P9.28 184 A12 e fast down gpio4_17
P9.42a 185 E14 e fast down gpio4_18
P8.10 186 A13 e fast down gpio6_4
P8.07 187 G14 e fast down gpio6_5
P8.08 188 F14 e fast down gpio6_6
P9.11a 203 B19 e fast down
P9.13a 204 C17 e fast down
P8.38b 205 C18 e fast down
P8.37b 206 A21 e fast down
P8.31b 207 G16 e fast down
P8.32b 208 D17 e fast down
P8.21 223 AD4 e fast down gpio6_29
P8.20 224 AC4 e fast down gpio6_30
P8.25 225 AC7 e fast down gpio6_31
P8.24 226 AC6 e fast down gpio7_0
P8.05 227 AC9 e fast down gpio7_1
P8.06 228 AC3 e fast down gpio7_2
P8.23 229 AC8 e fast down gpio1_22
P8.22 230 AD6 e fast down gpio1_23
P8.03 231 AB8 e fast down gpio1_24
P8.04 232 AB5 e fast down gpio1_25
P9.23 237 A22 e fast down gpio7_11
P9.22b 240 A26 e fast down gpio7_14
P9.21b 241 B22 e fast down gpio7_15
P9.18a 242 G17 e fast down gpio7_16
P9.17a 243 B24 e fast down gpio7_17
```

pdp7, 2020-05-29:
> Re: BeagleBone AI
> 
> by jgomes78 on Fri May 29, 2020 2:30 pm
> 
> Good Morning!
> I tried other GPIO's, but the following message appeared:
> 
> info: No pinmux for P9_15
> (node:2228) [INSPECTOR_ASYNC_STACK_TRACES_NOT_AVAILABLE] Warning: Warning: Async stack traces in debugger are not available on 32bit platforms. The feature is disabled.
> 
> info: No pinmux for P9_18
> error: Unable to write to /sys/class/gpio/gpio4/value
> (node:2298) [INSPECTOR_ASYNC_STACK_TRACES_NOT_AVAILABLE] Warning: Warning: Async stack traces in debugger are not available on 32bit platforms. The feature is disabled.
> error: Unable to write to /sys/class/gpio/gpio4/value
> 
> info: No pinmux for P8_6
> error: Unable to write to /sys/class/gpio/gpio35/value
> (node:2469) [INSPECTOR_ASYNC_STACK_TRACES_NOT_AVAILABLE] Warning: Warning: Async stack traces in debugger are not available on 32bit platforms. The feature is disabled.
> error: Unable to write to /sys/class/gpio/gpio35/value
> 
> 
> I also tried again on P9_15 and the same message appeared.
> The USR LEDs flashed normally.
> 
> Grateful.

MaBecker, 2021-11-12:
P9_15 is gpio76 on BBAI